open62541/open62541 #569: Connection issue on OpenWRT Atheros AR9330

Goran58, 2016-03-26:
Hi,
I have a major problem with connection to an open62541 server I have built for OpenWRT, Atheros AR9330.

When I try  to connect from UaExpert I get several decoding errors. Please see attached log file:
[uaexpert.txt](https://github.com/open62541/open62541/files/190386/uaexpert.txt)

Could It be an Little/Big endian issue? AR9330 is Big Endian.
:

jpfr, 2016-03-26:
Ok. I'll look into that.

Meanwhile, our work is a lot easier if you provide also
- the log output of open62541 (at best with the log threshold set to 100, so we see debug messages)
- a dump of the tcp connection (tcpdump, dumpcap and windump are tools that let you record network traffic)

Goran58, 2016-03-26:
@jpfr 
Thanx for your help. Attached are the log files.
[opc-server.txt](https://github.com/open62541/open62541/files/190433/opc-server.txt)
[tcpdump.txt](https://github.com/open62541/open62541/files/190434/tcpdump.txt)

I have used your code from Feb 4 22:57:19 2016. Only thing I have patched is to replace CLOCK_MONOTONIC_RAW with CLOCK_MONOTONIC.

jpfr, 2016-03-26:
The error code in the ua expert log hints at a problem with the message size.
http://open62541.org/doc/current/statuscodes.html
> #define UA_STATUSCODE_BADENCODINGLIMITSEXCEEDED 0x80080000 // The message encoding/decoding limits imposed by the stack have been exceeded.

Before we try to find an endianness issue, are you exchanging really long strings or data arrays?
In the tcpdump log, there are however no overly long messages...

Goran58, 2016-03-26:
I have exactly the same issue with the very simple **server_firstSteps**    

The problems started when I moved from Ubuntu64 to OpenWRT.

jpfr, 2016-03-28:
Hey @Goran58,

can you try the current master branch?
All endianness troubles should be resolved after f441d715430ded9c3fac38ea2c39c8eb99a2bf3b.

If you still get an error, please record a network dump with the actual message contents.
Your last post contained only meta-information like the message length.
If the network content is too sensitive fo public posting, you can also send me a private message.

Goran58, 2016-03-28:
@jpfr I have mailed log files to your "kit"-address.

Goran58, 2016-03-28:
@jpfr 
Couldn't the problem be that the BIG endian platform is not detected? What is the easiest way to force the code to BIG endian?

jpfr, 2016-03-29:
To check your endianness handling, you can insert #warning lines
into /include/ua_config.h.in (or directly in the generated
build/src_generated/config.h).

There, you can also force macro definitions.
They are all evaluated in /src/ua_types_encoding_binary.c

jpfr, 2016-04-03:
Commit be19c216483c121b85438af0cefd74b256f07797 should resolve the issue.
Awaiting confirmation.

closed 2016-04-04 by jpfr