open62541/open62541 #570: UA_CLIENTSTATE_ERRORED is not recoverable

ichrispa, 2016-03-29:
I encountered a slight problem I would like everyones opinion on how this should be handled.

Our client cannot decode the ServerStatus Datatype. Attempting to decide this value (or a few of the demo types such as DataType) will cause decodeBinary to fail. In this case, the error is on the client side - it is not inconceivable for other stacks to really send bad values though. In any case, this will also cause the client to go into the UA_CLIENTSTATE_ERRORED state.

There is no way to recover from this state - the connection is still there and the services will actually work properly. However services such as manuallySendPublish have become a bit discriminate and will return BADERRORNOTCONNECTED immediately (which is wrong). One can only disconnect, reinit and reconnect... all subscriptions and monitored items are in that case gone. There is also no way for userland to discern this error or assert the current client state.

I suggest to fix this error by having every successful service call actually set UA_CLIENTSTATE_CONNECTED. That would allow the client to recover from the error state.

What do you think?

jpfr, 2016-03-29:
I think it is a good idea to have `UA_CLIENTSTATE_ERRORED` to indicate when something is really borked and the client will not even attempt to repair anything.

Repairable states should get additional enums.
The standard lets use recover a session by switching from a broken securechannel.
And we could do some similar recovering internally when for example message sending failed during a short period (network congestion, lack of memory, etc.)

Stasik0, 2016-03-29:
I think its a good idea, actually errored state was intended to be used on errors that require a reconnection.

ichrispa, 2016-03-29:
Could we create something akin to UA_CLIENTSTATE_FAULTED and introduce a UA_Client_getState() to userspace? Right now only the stack internals know about the error, resulting in quite inconsistent behavior - highlevel will inexplicably fail without sending data, but calling most services still works.

Right now I have a (dirty) workaround in renewSecureConnection: If the client is ERRORED, but renewing the connection succeeds, I enforce a CONNECTED state. Basically, if I can still contact the server and the connection still exists, I overwrite the error. The same is done after each service that returns a good statuscode. Would a more orderly implementation of this workaround be suitable as a fix?

Stasik0, 2016-03-31:
IMHO yes

commit 8c22a486 (2016-04-04): Made client state visible to userspace; client recovers from potential decoding errors after successfull service calls or channel renewals (closes #570).
closed 2016-04-04 by ichrispa