open62541/open62541 #572: Subscriptions not working properly anymore

NetoLe, 2016-04-01:
Well I can't really step into the code in this case so I try to describe the problem.
My device is an ARM-Device with a server with the latest master-branch build, so is a PC-based programmed client. 
after the last changes in the subscriptions it stopped working correctly handling those subscriptions. 

The occuring is following: I didn't change my `UA_Client_Subscriptions_manuallySendPublishRequest` on the Client side which I used before the update, so it is called every 500 ms and now it doesn't get any value back. every PublishResponse is empty besides the value in the device really did change.

When i use UaExpert it worked before correctly but now, after the update, it once reads the value and then subscribes which works regarding the updated timestamps but it doesn't change the value once, means every PublishResponse gives the same value back, which is not different to the one before, but in the Server the value really did change. 

The next mystery is the NodeId. There is a MonitoredItem created for the numerical ID 11003 but the PublishResponse gives a value for the Node 811, which is in hex just two reversed bytes. see the screenshots for that:

[image]

[image]

And another mystery is: when I connect my self-programmed Client and after that connect UaExpert, UaExpert periodically updates the timestamp of the monitored Node. When I disconnect my Client UaExpert stops updating the timestamp which relys to empty PublishResponses. 
Besides that UaExpert still causes a Spamming of PublishRequests on the Wire.

Any Idea where I can start tracking the problem?

thanks so far

NetoLe, 2016-04-01:
small info: when I revert back to the version before https://github.com/open62541/open62541/commit/355d2574c00bf3e3244758e7bceb093ed22678d7 all works fine. So I guess it really relys on the new subscription changes.

jpfr, 2016-04-03:
> every PublishResponse is empty besides the value in the device really did change.

Sorry, I could understand that sentence in two ways. The PublishResponse is empty *if the value did not change* **or** empty *even though* the value changed.

> Well I searched through the traces and found this, where are two Notifications of the same node in one packet:

That's okay. One publish response can return multiple notifications if you have several Monitoreditems running. And one notification may contain multiple data samples if the sample frequency is higher than the publication frequency.

For me, subscriptions work fine with UAExpert. So I can't reproduce the error easily.
Can you provide a pcap log of the network traffic?

jpfr, 2016-04-04:
It seems our client is not playing nice with asynchronous requests where we queue publish responses and might send other responses first. Eveything is fine in UAExpert though.

I reverted some of the last commits in 8dd7c7e3255c314ad293d9e335e4fbc0ea9bc578.

Work will continue in this branch: https://github.com/open62541/open62541/tree/refactor_subscriptions
So that the changes will land for our server and client implementation simultaneously.

closed 2016-04-04 by jpfr

NetoLe, 2016-04-05:
it's fine now. your revert fixed the problem :) thanks!

and meant was:
>  empty even though the value changed.