open62541/open62541 #576: dynamic memory constantly increasing on Server-side with every new Client.

NetoLe, 2016-04-08:
I am using an embedded device with an ARM-architecture and my open62541-Stack is included in an C++-Framework. It is working so far. But there is a small restriction I experienced. 

I created a "debug-node" to which I can subscribe and which shows me the used dynamic memory-size (using tlsf for allocation etc.) and gets me an overview. I wrote a "debugging-client" with the open62541-stack too with which I can subscribe to that node and get me the used memory output if its changing (which works with UaExpert too very well now). 

And third I have another MFC-using-Client with the open62541-Stack which subscribes to all the other relevant data of the embedded device.

Then I saw something unusual. Everytime I connect to the server on the embedded device, the memory is increasing - ok, that's normal, monitored items are created etc. But when I disconnect it keeps an offset of ~200 Byte. So the more Clients I connect and disconnect, the more dynamic memory is used.

The memory pool itself is not a global pool but a pool just for the Stack-working stuff. I wrapped the networklayer-functions of your tcp_networklayer (great work btw) to functions, which I am using in the device, which implicates all the mallocs and frees are UA-mallocs etc. too, using the same memory-pool. 

And I am not using multithreading.

Stasik0, 2016-04-08:
Sounds like a memleak, are you able to run your your client in valgrind with leak-check=full to show where exactly its leaking?

NetoLe, 2016-04-12:
I can't run valgrind (using windows).

But I figured out, that the problem are the sockets. they are not free'd after disconnecting
`static void socket_close()` Isn't called once. I found this line: `/* only "shutdown" here. This triggers the select, where the socket is "closed" in the mainloop */´ which seems it doesn't for me.

NetoLe, 2016-04-12:
ok, this relates to https://github.com/open62541/open62541/issues/517 .

It seems the networklayer in my device does not work as expected. When the client sends a TCP with FIN,ACK my device just answers ACK which results in a RST, ACK after 2 min, but that doesn't reset the ports. I have to search the problem on my device.

closed 2016-04-12 by NetoLe