open62541/open62541 #578: OPC-UA Device Integration XML creates invalid C-code

truschival, 2016-04-11:
The XML-Namspace compiler ``generate_open62541CCode.py`` creates incorrect C-Code for the the xml specification Device Integration (DI) companion specification [https://opcfoundation.org/UADevices/1.1/Opc.Ua.Di.NodeSet2.xml] 

The XML defines nodes with  BrowseName ``1:&lt;MethodIdentifier&gt;`` and  ``<DisplayName>&lt;MethodIdentifier&gt;</DisplayName>``. (IMHO a quite shady cornercase) 
The script turns this into the macro `` #define UA_NS1ID_<METHODIDENTIFIER> 6018 `` which obviously is not correct C-Code.

jpfr, 2016-04-11:
Good find.
Imho we can just filter out any characters besides [a-z0-9_].

@ichrispa Any strong opinions in that regard?

Stasik0, 2016-04-11:
Or maybe be a bit paranoic and use the html encoded symbols like &gt; -> _GT_ or smth similar?

truschival, 2016-04-11:
@jpfr I would not just filter them out (the identifier could become empty) but substitute with another ASCII symbol
@Stasik0  Since it is not in visible to the (end-) user I don't think it is _necessary _to come up with a list of specific meaningful substitutions like ``_GT_``. The problem is we would need a (quite long) list for all kinds of UTF symbols that are AFAIK valid XML.... But you have a point, the resulting name would be unique. 
That said, the situation is not as easy as it looked.... a possible solution is for every new node, create a name like today, if it contains a "forbidden character" substitute the forbidden characters with a ``_`` if the resulting symbol is already in the list of known node substitute it with something else...?

truschival, 2016-04-12:
The developer will notice that the nodes have been renamed, I guess we should issue a warning during generation

Stasik0, 2016-04-12:
Maybe just encode all utf8? http://stackoverflow.com/a/2365444a

jpfr, 2016-04-12:
It's a starting point. `<` is still ASCII.

ichrispa, 2016-04-14:
I am not sure how to handle these "special" cases. The aliases are there to assist with the C Code and make it a bit more readable. Having weird unicode or C patterns in browseNames is a weird case.

I would just shrug and remove them... otherwise yes for filtering during the print (which will generate the next problem: someone will complain about duplicate names).

commit 08b3fb8f (2016-04-15): Generate nodemacros out of valid characters only (#581)

* Code generation: replace punctuation characters for macro identifiers

Fix #578. Replace punctuation characters with '_'

* Git commit id for out-of-tree builds

If the build directory is not a subdirectory of the source directory the
external command to read the GIT_COMMIT_ID failed.
The information was not available in the generated headers.
closed 2016-04-15 by jpfr