open62541/open62541 #579: Namespace generation does not link nodes

jpfr, 2016-04-13:
See this thread for background discussion:
https://groups.google.com/forum/#!msg/open62541/6s2zzakx82U/1VtJi_ioAwAJ

The input nodeset xml is as follows.
```xml
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:s1="http://festo.com/DI_test/Types.xsd" xmlns:s2="http://opcfoundation.org/UA/DI/Types.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema">

    <NamespaceUris>
        <Uri>http://festo.com/DI_test/</Uri>
        <Uri>http://opcfoundation.org/UA/DI/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="HasSubtype">i=45</Alias>
    </Aliases>
    <Extensions>
        <Extension>
            <ModelInfo Tool="UaModeler" Hash="IoOM2U7UwUN6RP1vvN3tkw==" Version="1.4.2"/>
        </Extension>
    </Extensions>
    <UAObjectType NodeId="ns=1;i=1502" BrowseName="1:AcmeDev">
        <DisplayName>AcmeDev</DisplayName>
        <Description>Any device by acme company</Description>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">ns=2;i=1002</Reference>
        </References>
    </UAObjectType>
</UANodeSet>
```

The nodeset generator ran as
`python pyUANamespace/generate_open62541CCode.py schema/namespace0/Opc.Ua.NodeSet2.xml -i pyUANamespace/NodeID_Blacklist_FullNS0.txt input.xml myNamespace`

Out comes the following C-code:
```c
UA_INLINE void myNamespace(UA_Server *server) {
    do {
        // Node: opcua_node_objectType_t(ns=1;i=1502), 1:AcmeDev
        UA_ObjectTypeNode *opcua_node_objecttype_t_ns_1_i_1502 = UA_NodeStore_newObjectTypeNode();
        opcua_node_objecttype_t_ns_1_i_1502->browseName = UA_QUALIFIEDNAME_ALLOC(1, "AcmeDev");
        opcua_node_objecttype_t_ns_1_i_1502->displayName = UA_LOCALIZEDTEXT_ALLOC("en_US", "AcmeDev");
        opcua_node_objecttype_t_ns_1_i_1502->description = UA_LOCALIZEDTEXT_ALLOC("en_US", "Any device by acme company");
        opcua_node_objecttype_t_ns_1_i_1502->nodeId.namespaceIndex = 1;
        opcua_node_objecttype_t_ns_1_i_1502->nodeId.identifier.numeric = 1502;
        // Parent node does not exist yet. This node will be bootstrapped and linked later.
        UA_RCU_LOCK();
        UA_NodeStore_insert(server->nodestore, (UA_Node*) opcua_node_objecttype_t_ns_1_i_1502);
        UA_RCU_UNLOCK();
    } while(0);
}
```

What should happen is that the generator assume that all nodes in the ignore-blacklist already exist on the server. It does however not create the *HasSubtype* reference from the parent.

@ichrispa 
You're the original developer of the namespace generation. Any pointers to start debugging?

truschival, 2016-04-14:
Thanks for investigating and verifying the issue. I would like to point out that it seems there is more to the issue  than  missing _HasSubtype_ reference.  
To me it seems that all nodes are generated in namespace index 1 (or whatever they are defined in the individual xml files)  hence we have collisions between nodeids. While the namespaces are unique and the nodeids are unique in their respective namespace the indices of the namespaces are not unique. 
Replacing the  nodeid ``1500`` of ` <UAObjectType NodeId="ns=1;i=1502" BrowseName="1:AcmeDev">` 
with ``1002``  will result in model that apparently misses DeviceType (namespace OPC.UA.DI). 

In the above model you instantiate an object of type ``AcmeDev`` as component of the Node ``DeviceSet`` it will not appear show. however it you can find the Object in the ObjectType Hierarchy under ``TopologyElement``. (ParameterSet  id5002 dissappears because of nodeid clash)

The attached XML model shows the issues. To avoid issue #578 use the attached OPC_UA_DI_Nodeset.xml
[Opc_Ua_Di_NodeSet2.xml.txt](https://github.com/open62541/open62541/files/218529/Opc_Ua_Di_NodeSet2.xml.txt)
[di_test.xml.txt](https://github.com/open62541/open62541/files/218530/di_test.xml.txt)

ichrispa, 2016-04-14:
I am not only the "original creator of", I still consider it my code @jpfr. 

The namespace compiler does not change nodeIds - this has been repeatedly tested and verified. Check the data model. On request of stasik, the compiler will overwrite nodes if duplicate ids are found - which would in theory allow to overwrite a default definition with a more precise model. This behavior is configurable.

Ignored nodeIds are added to the printedNodes list. The actual node definition must exist and must have been parsed before the child nodes are printed. This is a compiler, not a model transformer. It relies on parsing and understanding the information models.

Solution: Feed the compiler a valid XML containing the nodes that are to be ignored as part of a valid data model. The actual order of the files should not matter, linking is done after parsing. 

If that does not work: It might well be possible that the parent node does not exist yet. When compiler checks if the node exists before linking. If the node is printed later, the reference will be created when the parent is instantiated. Check the parent node's instantiation and see if the compiler has added the missing node there.

ichrispa, 2016-04-14:
In regard to the compilation aspect: When you link a binary, the compiler still checks the symbols existence prior to linking... same thing here: The compiler checks for the nodes existence prior to linking - even if that node is not printed itself.

So in summary: Always feed the namespace compiler the entire model, even if you only want to print part of it.

ichrispa, 2016-04-15:
Verified to be wrong compiler use and not a bug. Use:

> python pyUANamespace/generate_open62541CCode.py schema/namespace0/Opc.Ua.NodeSet2.xml -i pyUANamespace/NodeID_Blacklist_FullNS0.txt Opc.Ua.NodeSet2.xml input.xml myNamespace

closed 2016-04-15 by ichrispa

truschival, 2016-04-19:
The hint does not work for me, I don't know what I am doing wrong, but the relationships are not created. I am inclined to reopen the issue.

Please verify my compiler command line for userspace compilation (as printed by `make VERBOSE=1
[di_test.txt](https://github.com/open62541/open62541/files/225471/di_test.txt)
`). 
`
python.exe J:/projects/open62541/tools/pyUANamespace/generate_open62541CCode.py  
-i J:/projects/open62541/tools/pyUANamespace/NodeID_Blacklist_FullNS0.txt 
J:/projects/open62541/tools/schema/namespace0/Opc.Ua.NodeSet2.xml 
J:/projects/open62541/tools/schema/namespaces/Opc_Ua_Di_NodeSet2.xml 
J:/projects/open62541/examples/DI_Test/di_test.xml 
C:/tmp/open62541_gcc/src_generated/nodeset 
`

The fix for issue #578 and #580 do work. Modeling with "Unified-Automations UaModeler" I made sure all nodeids of my model are globally unique staring at ``21000``. Neither ``Opc.Ua.NodeSet2.xml`` nor ``Opc_Ua_Di_NodeSet2.xml`` contain any nodeids above ``21000``. Hence even if the namespaces are all mapped to nsid=1 there is no collision. I ran consistency checks and found no errors.
Still the relationships are not created. I can't find the subtype ``AcmeDev`` below ``BaseObjectType/TopologyElement/DeviceType``. Nor can I find an Instance of ``AcmeDev`` below ``Objects/DeviceSet``. Looking at the generated ``nodeset.c`` there are no references to AcmeDev.

See the input XML below: 
[di_test.txt](https://github.com/open62541/open62541/files/225472/di_test.txt)