open62541/open62541 #580: C-Macro-names in generated nodeset.h are not unique

truschival, 2016-04-14:
It seems the OPC.UA Device Integration Companion spec [https://opcfoundation.org/UADevices/1.1/Opc.Ua.Di.NodeSet2.xml]  is the killer-testcase for code-generator (related to #578 and #579).

The code generator does not take parent nodes into account when generating C-Code identifiers which leads to the fact that one symbolic name refers to two different nodeids. 

Compilation with gcc 4.8.0 and the default flag ``-Werror`` fails for the ``nodeset.h`` generated from the above XML with the error of redefined macros. (e.g. ``UA_NS1ID_METHODIDENTIFIER_`` = nodeid ``6029`` and  ``6018``). 
This issue is caused by the fact that 2 different nodes have the same name and _BrowseName _. This is legal because the two nodes are in entirely different parts of the model and have different parents.
See line 141 and 727 in the aforementioned xml model.

ichrispa, 2016-04-15:
The given feature of the namespace compiler is a convenience. The UA_NSxID_browseName identifiers are used throughout open62541 and in keeping with this naming scheme types found in other namespaces are named accordingly. The idea is that one propably knows how the node is named, but not the by necessity its ID - also makes code more readable.

The problem that two nodes in the same namespace may have the same browsename has simply not been encountered before. Frankly, I think that its bad practice, but since its the opcfoundations bad practice, it makes it good bad practice...

Hence the question: How should this be handled in an intuitive manner? Change the entire naming scheme? Don't generate duplicates (which will result in the next issure being named "ns compiler does not generate alias for node XY")? Append a counter (_1, _2,...) or the nodes numeric id (what if it doesn't have a numeric id)?

@truschival: As a temp fix, just disable the feature by commenting tools/pyUANamespace/open62541_MacroHelper.py#L77

ichrispa, 2016-04-15:
Post-Mortem: I picked the counter-solution... anyone with a better idea is welcome to propose it.

@truschival: Since I suspect where you are going with the assorted namespace-compiler issues, you might want to modify the CMakeFile#311 to this:
```
                                                ${PROJECT_SOURCE_DIR}/tools/schema/namespace0/${GENERATE_NAMESPACE0_FILE}
                                                ${PROJECT_SOURCE_DIR}/tools/schema/Opc.Ua.Di.NodeSet2.xml
                                                ${PROJECT_BINARY_DIR}/src_generated/ua_namespaceinit_generated
```
 to enable automatic compilation of the Di model.

If you insist on userspace compilation, I suggest you add UA_EXPORT to encodeBinary (methods in user namespace can otherwise not be encoded).

closed 2016-04-15 by ichrispa

truschival, 2016-04-18:
> The given feature of the namespace compiler is a convenience. 

This makes perfect sense and is a good idea. 

> The problem that two nodes in the same namespace may have the same browsename has simply not been encountered before. Frankly, I think that its bad practice, but since its the opcfoundations bad practice, it makes it good bad practice...

OPC UA Spec Part 3 Section 5.4.2 confirms this: 
> Unlike NodeIds, the BrowseName cannot be used to unambiguously identify a Node. Different
Nodes may have the same BrowseName.

AFAIK the  BrowseName  of a node is the part it adds to the "browsepath" much like a folder or file in a filesystem.  

> Hence the question: How should this be handled in an intuitive manner? 

Maybe we could add browsenames of the parent nodes recursively until the identifier becomes unique. What do you think?

commit d1ccf6fc (2016-04-15): Typedefinitions are checked for uniqueness; non-unique names are autoincremented (closes #580)