open62541/open62541 #583: Put client connection into the config (change API before 0.2?)

jpfr, 2016-04-18:
Hey there,

According to issue #567, I'd like to get comments on whether to make this breaking change before 0.2:

All the client connectivity functions take a connection function pointer as input.
```c
typedef UA_Connection (*UA_ConnectClientConnection)(UA_ConnectionConfig localConf,
                                                    const char *endpointUrl,
                                                    UA_Logger logger);

UA_StatusCode UA_EXPORT
UA_Client_connect(UA_Client *client, UA_ConnectClientConnection connFunc, const char *endpointUrl);
```

I'd like to put the connFunc pointer into the client config instead.
The two main reasons for this are:
- Most people will only use the standard TCP implementation.
So this pointer is a messy detail these users should not be forced to deal with.
- There is little reason to change the connection pointer between connections. It seems more logical to attach it to the client.

HolgerJeromin, 2016-04-18:
Looks good to me

FlorianPalm, 2016-04-19:
I am good with it. I think there is more to fix within the connection structure. To me it is quite confusing that we manage a localConf and a remoteConf (sendBufferSize/recvBufferSize on both sides).

jpfr, 2016-04-19:
Related PR #585.

ichrispa, 2016-04-19:
Yes, please make the connFunc vanish from connect - I think its counter-intuitiv in this context...

jpfr, 2016-04-19:
#585 merged.

closed 2016-04-19 by jpfr