open62541/open62541 #584: Support of constitent namespace ids

truschival, 2016-04-19:
This issue is opened as clean start for the discussion in #579 [https://github.com/open62541/open62541/issues/579#issuecomment-209781840]. 
I have the impression the namespace generator does not treat namespace indices (``ns=``) correctly when compiling several XML input files.

The ``ns=`` is only unique in the context of one XML-File and there is no guarantee that the order (i.e. ``ns=``) of a namespace-uris in a XML file remains the same. Only NS0 is fixed.
E.g. in ``Opc_Ua_Di_NodeSet2.xml`` the URI ``http://opcfoundation.org/UA/DI/`` corresponds to ``ns=1``, there is no other. In other files, models referencing it could be ``ns=2`` or ``ns=3``.
The nodeids and namespace ids have to be interpreted locally to the XML file when compiling the model.
We have to make sure that ``ns=1;12345`` in the first xml-file does not collide with ``ns=1;12345`` in another xml-file where the namespace-uri-list is different.

I didn't dig in deeply into the generation script yet but I think we need something like the following approach:

1.  When reading any new xml file, check the namespace-uri list and add all yet unknown uris to global (script internal list). 
2.  For the context of this xml file interpret namespace ids locally, i.e. check what uri they correspond to.
3.  replace the local nsid with their corresponding index in global list.

Any comments and hits on this approach are welcome. I have not yet put in effort on how to treat blacklisting.

Thanks in advance!

ichrispa, 2016-04-19:
Hi @truschival ,

thank you for bringing the issue to our attention, but with all due respect: "we need to" is the wrong qualifier here (no offense intented). 

The fix you propose seems reasonable, I am however not sure if it is correct from the stack to assume the right to change namespaces in this case. I consider namespace collisions to be a userspace/modelling problem, as the namespace compiler does exactly what the XML files request. Having the compiler start to "repair" problematic models is - for me at least - not an option.

```
The ns= is only unique in the context of one XML-File and there is no guarantee that the order (i.e. ns=) of a namespace-uris in a XML file remains the same. Only NS0 is fixed.
E.g. in Opc_Ua_Di_NodeSet2.xml the URI http://opcfoundation.org/UA/DI/ corresponds to ns=1, there is no other. In other files, models referencing it could be ns=2 or ns=3.
The nodeids and namespace ids have to be interpreted locally to the XML file when compiling the model.
```
Is this convention documented? I can't find it in the spec, but if it is a guideline relating to 62541, it makes the course of action very straigtforward. If it is not part of the spec, then the namespace compiler is not at all at fault here.

Replacing the namespace index should be feasible. However, how should the following scenario be dealt with: 
* The user creates two namespace XMLs using NS=1 and NS=2 respectively
* The namespace 2 xml references nodes from namespace 1, as the user somehow coupled the information models
* The user then adds something like the DA XML to the mix.
* The users namespace 1 XML gets changed to ns=3 as it collides with the DA XML
* The users namespace 2 XML becomes invalid...
This scenario would assume that the namespace compilation process has already linked the nodes of the users ns1 and ns2 prior to parsing the DA XML... or we need to introduce some sort of parsing priority order like `((Opc.Ua.NodeSet2.xml (usersNs1.xml userNs2.xml)) OpcDa.xml)` to signal that some namespace are coupled and interdependant.

Best regards,
ichrispa

There is a more than simple hot fix: `sed -i 's/ns=1/ns=2/g' faultyNamespace.xml`

jpfr, 2016-04-20:
To me, @truschival makes a good case.
I think we can solve the problem and still allow that address space xmls are compiled and loaded independently.

The standard requires that in the namespace array
0 -> “http://opcfoundation.org/UA/”
1 -> URI of the application

> Part 5: 6.3.1 The URI of the ServerArray with Index 0 shall be identical to the URI of the NamespaceArray with Index 1, since both represent the local Server.

So ns0 and ns1 are fixed after startup of the server. Afterwards, "first come first serve" seems to be the only feasible approach. The next namespace URI gets mapped to 2, then 3, and so on.

Take the following example: 
```xml
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:s1="http://festo.com/DI_test/Types.xsd" xmlns:s2="http://opcfoundation.org/UA/DI/Types.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://festo.com/DI_test/</Uri>
        <Uri>http://opcfoundation.org/UA/DI/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="HasSubtype">i=45</Alias>
    </Aliases>
    <Extensions>
        <Extension>
            <ModelInfo Tool="UaModeler" Hash="IoOM2U7UwUN6RP1vvN3tkw==" Version="1.4.2"/>
        </Extension>
    </Extensions>
    <UAObjectType NodeId="ns=1;i=1502" BrowseName="1:AcmeDev">
        <DisplayName>AcmeDev</DisplayName>
        <Description>Any device by acme company</Description>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">ns=2;i=1002</Reference>
        </References>
    </UAObjectType>
</UANodeSet>
```

We could assume that ns1 in the XML file refers to the first URI in the NamespaceUris list, i.e. "http://festo.com/DI_test/". Then it is easy to defer the mapping of the namespaces **at runtime** when the generated C-code is executed.
1. Check if the URIs used in the XML are alrady there or add them to the namespacearray of the server. Then write the mapped URI id into a variable.
```c
int xml_ns1 = UA_Server_addNamespace(server, "http://festo.com/DI_test");
int xml_ns2 = UA_Server_addNamespace(server, "http://opcfoundation.org/UA/DI/");
```
2. Add all nodes with reference to the mapping variables and not hard-coded identifiers.

The first step would be to adapt `UA_Server_addNamespace` so that it returns the id of an existing namespace if the URI already exists. That would be 2 lines of code....

truschival, 2016-04-20:
Thanks for the discussion. Checking the literature it my assumption is not explicitly documented.
The only thing that is explicitly documented is the uniqueness of a node within its namespace uri. (OPC-UA Spec Part 3 section 8.2.2 NamespaceIndex). 
I couldn't find any documentation on the uniqueness of namespace indices, except for the quote @jpfr mentioned. Rather the opposite (at least during runtime): 
"The NamespaceIndex is introduced for optimization purposes [...] The NamespaceIndex is a pointer to the NamespaceArray of the server [...] when a client disconnects form a server it does not have any guarantees on the NamespaceArray. Thus the namespace URI represened by index '3' could be represented by index '5' when the client reconnects."(Wolfgang Mahnke et. a. OPC Unified Automation, Springer 2009 Chapter 2.8.5) - Admittedly this is not a IEC standard

The OPC Unified Architecture for Analyser Devices (ADI) [https://opcfoundation.org/UA/schemas/ADI/1.00/Opc.Ua.Adi.NodeSet2.xml] is modeled with namespace indices that correspond to the simple example above. 
`<NamespaceUris><Uri>http://opcfoundation.org/UA/ADI/</Uri><Uri>http://opcfoundation.org/UA/DI/</Uri></NamespaceUris>
`
Here nodes form DI are also referenced as ``ns=2``. Feeding this document together with the DI model to the the python script probably results in the same issue (not confirmed). 

I am not sure if I understood the scenario correctly. IMHO the xml itself never becomes invalid but only the mapping of indices to URIs is rearranged. I think the scenario is: 
1. user creates two xml files with namespace URIs  ``foo`` and ``bar`` each in a individual xml file, while ``bar`` references elements from ``foo``. Whithin each XML file the namespace array only has one entry, which in the first case contains ``foo`` as ns=1 and in the second case ``bar`` as ns=1 and ``foo`` as ``ns=2``. (like the ADI spec)
2. Another user builds a model ``baz`` that refers to ``bar`` and  implicitly ``foo``.  ``baz`` is ns=1 and in the users model ``foo`` can have ``ns=2`` or ``ns=3`` and bar the other. Nodes in ``baz`` will refer to other nodes form ``foo`` and  ``ns=2;12334```. 

What am I missing?

ichrispa, 2016-04-20:
You are assuming that the namespace array is referenced within nodeids... which it is not. Only extendedNodeIds can reference URIs, and those are not part of references by default. Given that ns=1;i=12345 is the only linkable information provided by a node, it can not be determined which information model you assume is hiding behind the number "1". In order to reassign a namespace, the nodes need to be already linked, which obviously cannot be done until everything is parsed.

In your example: baz does not refer to foo or baz. It refers to a namespace "2". I have not yet seen an XML that contains information like "within this XML: referring to http://opcfoundation.org/UA/DI/ as 2, http://someorganisation/modelname as 4" that would allow a pre-linkage derefencing of a namespace number to a model. Do you have a model that contains this information? I can't even provoke UA Modeler to spit out this sort of information into an XML file.

As long as the  namespaces foo and bar each contain an isolated model, which is only instantiated during runtime, your strategy will work perfectly.

But let's assume you create a model based on the DA XML and you reference some types there. You call your namespace "2", DA is "1", OPC UA is still "0" - again, everything is peachy. The next corporation/user/university has another vendor model, which is also called "1" (which is kind of your alike to problem). That model is only dependant on namespace 0. Not we throw all three of them into the mix...

Whether your model "2" will still work is dependant on which namespace "1" gets reassigned the new namespace first. If the compiler picks the "other" corporation/user/university XML for renaming, you model will still work. If DA is changed to ns "3", then your model won't work, because you used namespace "1" to create namespace "2" under the assumption that "1" is the DA model. A specific precedence order must be kept, which depending on how many models are in the mix is not trivial.

The matter of fact is that this is not a compiler issue... it is a preprocessing issue. We do not have and do not claim to distribute a namespace preprocessing engine at the moment (though undouptedly, one could be written).

ichrispa, 2016-04-21:
In retrospect: There is a severe difference here between adding a namespace to a running server instance (which the document references) and creating a static initialization routine, which must present a coherent nodeset.

ichrispa, 2016-04-21:
@Stasik0: Seeing the line of commits (435498a) we are handling this issue at server runtime when adding new nodes to the nodestore? If so, can this issue be closed as not compiler related?

Stasik0, 2016-04-21:
The generated code has to be adapted, hasnt it?

ichrispa, 2016-04-21:
Since I'm not in on what you guys are up to, I'm not sure. I see the following options:

1) We could have the server add these nodes to a seperate nodeset, then attempt to "merge" them into the nodestore. The server has namespace URIs and Endpoints available at runtime to solve this issue. This is what I suspect the afformentioned commits are working towards. This solution would also cover the spec-quote from @truschival 's (server is able to change a node's namespace at runtime if required).

2) The namespace compiler can attempt to preprocess XML's. But without Namespace URI's to nsId descriptions in the XML files, the compiler can only guess what the user intents (the information is just not there). I highly suspect that this will lead to the next ticket being called "Namespace Compiler invalidates interdependant models by changing the namespace Id".

truschival, 2016-04-22:
@Stasik0 I would prefer not to close the issue as long as the discussion is ongoing.
I agree with @ichrispa, there is a severe difference between adding namespaces during runtime and at compile time. I also share the impression that there may be filed a ticket where another user complains about changing of namespaceids.
Nevertheless there is the question on how to deal with (compile-time) models with several xml-files containing namespace-uris in different orders. Which is unfortunate but a real case (see the Analyser Device XML-Model) 
The runtime approach would work and work consistently if the information from the xml-models was completely available at runtime.
Currently I am working on a solution, implementing the approach outlined above [https://github.com/open62541/open62541/issues/584#issue-149391447]. However I am not yet sure how to deal with the "reserved namespace index 1 for the local server".

uleon, 2016-04-22:
I'm very much in favor of compilers that do not guess what I really wanted in case of ambiguities. Therefore I'd like to see a simple and flexible preprocessor approach where users define the intended behaviour. I guess something like a configuration file with sections might come in handy, for instance 
``
[namespaces]
ns0=http://opcfoundation.org/UA/
di=http://opcfoundation.org/UA/DI
adi=https://opcfoundation.org/UA/schemas/ADI/1.00/Opc.Ua.Adi.NodeSet2.xml
myIM=http://myorganisation/modelname
yourIM=http://yourorganisation/modelname
omg=http://myorganisation/modelname

[myIM]
1=di
2=adi

[yourIM]
2=adi

[omg]
3=di
1=myIM
``
Based on this information the preprocessor would generate a conflict free namespace numbering and issue a sequence of sed scripts on the files before feeding them to the compiler. ( Please note that I do not care about the notational style  of such a preprocessor configuration file, could be anything from jason to xml )

uleon, 2016-04-22:
This preprocessor might be responsible for the coding issues as well, that is substituting all the characters that would lead to uncompilable C-identifiers. Eventually I would be happy if this preprocessor had a feature to handle the annoying uax:-prefixes as well .

ichrispa, 2016-04-22:
@truschival I am working on a preprocessor solution for the given issue. Can you maybe provide a usecase I can test against? All I currently have are dummy XMLs and the DI XML, which might not cover your usecase reliably...

truschival, 2016-04-22:
@ichrispa - cool, thanks for the efforts. 
Sorry, so far I also have been struggling with my tiny dummy models and have no bigger model at hand. 
BTW, I stopped using the Analiser Device Model (ADI) for first tests, as it has its own set gotchas, e.g. Data with double quotes within strings....

ichrispa, 2016-04-22:
OK gentlemen... not quite the day off I imagined, but fun non the less.

The preprocessor is up (passes tests on windows/linux) - I suggest someone run it against python 3 in the near future though. Here's how it works:

 * If more then one XML is passed, they first get assigned new incremental nsIds. Just about any every model will call itself "1" and use others in varying order (i.e. DI would be "2", PLCOpen "3", etc.). The order of the namespaces is not guaranteed, so DI might be "2" in modelA and "4" in modelB.
 * The Preprocessor attempts to relink the models. To that extend, it will attempt to use two methods in the following order:
  * If the XML defines "xmlns:s1, xmlns:s2..." URIs, those will be used to determine what that model refers to by the numeric id.
  * If those attributes do not exist, the <NamespaceUris> section will be used - the Preprocessor assumes that the URIs are ordered in the order the indexes are used. 
  * In both cases the preprocessor will now search for the document that has that URI (referenced as ns 1 in the document). 
  * If no URI could be determined for the target namespace index in the source document _or_ no document actually has that URI, the preprocessor will guess - yes, it does emit a warning. All references in the document in question to namespace X will be gathered. They are then looked up in all other documents (after tampering with the nsindex). The document that contains most of those reference target is assumed to be the correct namespace.
  * If that fails too, an Error will be emitted. The compiler will remove any reference that is not linkable and any node that is unreachable.

Apparently, XML DOM2 defines that values may be prefixed using the namespace qualifier - which explains the <uax:Argument> and the like being created by UA Modeller. The Compiler is a bit more accurate here - it want aliases to match the definition. To that extend, any namespace qualifiers, including the uax:, will be removed from the document.

Any preprocessor changes are printed into temprorary files (that are removed after parsing them in the compiler). The compiler gets fed the temprorary files, not the originals.

Feedback would be welcome.

truschival, 2016-04-25:
The implementation of @ichrispa resolves the issue for me, thanks.

As for larger "real-world" models as possible test-cases: there is a nicely documented [OPC Unified Architecture for AutomationML example](https://opcfoundation.org/developer-tools/specifications-unified-architecture/opc-unified-architecture-for-automationml/) by Fraunhofer IOSB including 2 large library namespaces and a small application. Maybe it makes a good test case for the futer.

closed 2016-04-26 by ichrispa