open62541/open62541 #594: QNX compilation

tpiat, 2016-04-22:
I have copmpilte open62541 under QNX
Two error
1. If file gulliver.h included many errors appear .
2. CLOCK_MONOTONIC_RAW is undefined.
    CLOCK_MONOTONIC defined in time.h (if defined __EXT_POSIX1_200112)

In file ua_types.c
Instead 
```
#ifdef __CYGWIN__    
    clock_gettime(CLOCK_MONOTONIC, &ts);
#else
    clock_gettime(CLOCK_MONOTONIC_RAW, &ts);
#endif
```
must be 
```
#if defined(__CYGWIN__) || defined(__QNX__) || defined(__QNXNTO__) 
    clock_gettime(CLOCK_MONOTONIC, &ts);
#else
    clock_gettime(CLOCK_MONOTONIC_RAW, &ts);
#endif
```
and remove #include <gulliver.h>

tpiat, 2016-04-22:
If gilliver.h included:

```
/usr/qnx650/target/qnx6/usr/include/gulliver.h:265: error: previous declaration of 'void __swap_sigset(sigset_t*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:265: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:266: error: previous declaration of 'void __swap_sigevent(sigevent*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:266: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:267: error: previous declaration of 'void __swap_siginfo(_siginfo*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:267: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:268: error: previous declaration of 'void __swap_timerinfo(_timer_info*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:268: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:271: error: previous declaration of 'void __swap_procfs_info(_debug_process_info*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:271: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:272: error: previous declaration of 'void __swap_procfs_mapinfo(_procfs_map_info*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:272: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:273: error: previous declaration of 'void __swap_procfs_status(_debug_thread_info*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:273: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:274: error: previous declaration of 'void __swap_procfs_channel(_debug_channel*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:274: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:275: error: previous declaration of 'void __swap_procfs_irq(_debug_irq*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:275: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:276: error: previous declaration of 'void __swap_procfs_timer(_debug_timer*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:276: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:277: error: previous declaration of 'void __swap_procfs_threadctl(_procfs_threadctl*, int)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:277: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:278: error: previous declaration of 'void __swap_procfs_sysinfo(syspage_entry*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:278: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:279: error: previous declaration of 'void __swap_procfs_debuginfo(_procfs_debug_info*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:279: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:280: error: previous declaration of 'void __swap_procfs_thread_name(_thread_name*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:280: error: conflicts with new declaration with 'C' linkage
```

tpiat, 2016-04-24:
I have next problem:
I write С/С++ program, QNX.
File open62541.h included both to main.cpp (C++)
and to open62541.c.

In the result 
/usr/qnx650/target/qnx6/usr/include/gulliver.h:265: error: previous declaration of 'void __swap_sigset(sigset_t*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:265: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:266: error: previous declaration of 'void __swap_sigevent(sigevent*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:266: error: conflicts with new declaration with 'C' linkage
 
The function prototype in main.cpp is "С++", in open62541 -- "C".
Can I solve this problem?

Wited extern "C" -- do not works.
#ifdef __cplusplus
extern "C" {
#endif
#include <gulliver.h>
#ifdef __cplusplus
}
#endif

When using "С" in files  -- works.

#ifndef __cplusplus
{

#include <gulliver.h>
} 

Can you insert one fix to ua_config.h? Need exclude  gulliver.h in   "C++"?
 __cplusplus or predefined constnt ?

```
#ifndef _DO_NOT_INCLUDE_GULLIVER_H
#include <gulliver.h>
#endif
```

Кстати, вообще без включения этого файла лично у меня вполне успешно компилируется.

jpfr, 2016-04-24:
Hey @tpiat

Can you try the current master?
This commit should fix the C++ interop issues you described: c53990382171843bb51e275cfbf50eb51bccf878

tpiat, 2016-04-25:
@jpfr 
```
/usr/qnx650/target/qnx6/usr/include/gulliver.h:265: error: previous declaration of 'void __swap_sigset(sigset_t*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:265: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:266: error: previous declaration of 'void __swap_sigevent(sigevent*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:266: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:267: error: previous declaration of 'void __swap_siginfo(_siginfo*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:267: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:268: error: previous declaration of 'void __swap_timerinfo(_timer_info*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:268: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:271: error: previous declaration of 'void __swap_procfs_info(_debug_process_info*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:271: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:272: error: previous declaration of 'void __swap_procfs_mapinfo(_procfs_map_info*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:272: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:273: error: previous declaration of 'void __swap_procfs_status(_debug_thread_info*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:273: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:274: error: previous declaration of 'void __swap_procfs_channel(_debug_channel*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:274: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:275: error: previous declaration of 'void __swap_procfs_irq(_debug_irq*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:275: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:276: error: previous declaration of 'void __swap_procfs_timer(_debug_timer*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:276: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:277: error: previous declaration of 'void __swap_procfs_threadctl(_procfs_threadctl*, int)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:277: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:278: error: previous declaration of 'void __swap_procfs_sysinfo(syspage_entry*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:278: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:279: error: previous declaration of 'void __swap_procfs_debuginfo(_procfs_debug_info*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:279: error: conflicts with new declaration with 'C' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:280: error: previous declaration of 'void __swap_procfs_thread_name(_thread_name*)' with 'C++' linkage
/usr/qnx650/target/qnx6/usr/include/gulliver.h:280: error: conflicts with new declaration with 'C' linkage
```

tpiat, 2016-04-25:
@jpfr 

I think there are internal QNX problem.
In system file netbyte.h are next lines
```
#ifndef _GULLIVER_H_INCLUDED
#include <gulliver.h>
#endif
```
Probably insert this lines to ua_config?

tpiat, 2016-04-25:
@jpfr 
Thank you for you thought.
I solve this problem.
QNX compiler does not work well with nested include. (Works incorrectly)
Now my clinent works with version 1cb82777b5 works without any chnages in open62541.

closed 2016-04-25 by tpiat