open62541/open62541 #595: UaExpert complains about invalid datatype for NodeClass

NetoLe, 2016-04-26:
Hello,

since the latest updates I have a problem which is that UaExpert complains about the Type of the NodeClass.

[image]
that doesn't influence the functionality of the server but seems a bit not correct.

Kruno313, 2016-04-26:
Same here.
I build it for the first time and tought it's normal.
Also amalgamated build do not work, and the demo server.c uses UA_ServerNetworkLayerTCP() with thre parameters. Had to make changes in it and build and against .so library.

Stasik0, 2016-04-26:
I though, 1cb82777b578f3273d35251b2552e88c288955d9 was supposed to fix it?

Kruno313, 2016-04-26:
I'm not sure what is affected with that change.
I pulled my source with 'git clone https://github.com/acplt/open62541'  as
described here:
http://open62541.org/doc/current/tutorial_server_firstSteps.html#prerequisites
and compiled it as described in the next chapter of the same link.
The demo server source in
http://open62541.org/doc/current/tutorial_server_firstSteps.html#creating-your-first-server
does not compile without two small changes and than, it has this issue.

Stasik0, 2016-04-26:
when did you pull it? which version of uaexpert do you use?

Stasik0, 2016-04-26:
okay, I can confirm it

Kruno313, 2016-04-26:
I pulled it last week, thursday or friday.
I'l try it again.

Kruno313, 2016-04-26:
It works fine now!
Just buildt it fresh and it's perfect.
Thanks

2016-04-26 15:21 GMT+02:00 krunoslav ostrouska <
krunoslav.ostrouska@gmail.com>:

Stasik0, 2016-04-26:
@NetoLe can you confirm it?

Stasik0, 2016-04-26:
@Kruno313 nice to hear it

ichrispa, 2016-04-28:
I can also confirm that @jpfrs fix has corrected this problem.

closed 2016-04-28 by ichrispa

NetoLe, 2016-04-29:
confirmed. thanks :)