open62541/open62541 #596: Server failis to handle tcp frames

Stasik0, 2016-04-26:
Hey @open62541/open62541-core,

Just testing around a bit I have found a following bug. While connecting to our server from uaexpert from vm-brige, the createsession request gets fragmented. The server fails to decode the second chunk (i think) and hangs completely. In the wireshark log you will see that it ignodes new hello messages from the client. 

Looking around in the code, I did not find the network-layer buffer that had the task of collecting tcp frames to complete messages. I believe that server worker gets a job with a partial messages. 2 questions: - did the buffer go intentionally? - why does the server hang?

Here is the console log:
```
[04/26/2016 14:19:15.369] info/network	TCP network layer listening on opc.tcp://*.45:16664
[04/26/2016 14:19:31.425] info/network	New Connection 5 over TCP from *.82:55948
[04/26/2016 14:19:31.429] info/network	Decoding of message header failed on Connection 5
[04/26/2016 14:19:31.429] info/network	Closing the Connection 5
^C[04/26/2016 14:19:56.952] info/server	Received Ctrl-C
[04/26/2016 14:19:56.952] info/network	Shutting down the TCP network layer with 1 open connection(s)
```

and the wireshark log
[tcp_frames_fail.pcapng.zip](https://github.com/open62541/open62541/files/236796/tcp_frames_fail.pcapng.zip)

jpfr, 2016-04-26:
This is the function that returns only complete chunks and buffers half-received chunks.
https://github.com/open62541/open62541/blob/master/src/ua_connection.c#L33

ichrispa, 2016-04-26:
I'm not really qualified to diagnose this bit - haven't touched that code in ages and it has evolved quite a bit. I'll just offer thoughts.

Judging from the packet capture, this is not a regular TCP reset condition. If this is what went over the wire, both HEL messages are part of the previously established connection and they represent valid data. I suspect the server has received a partial MSG message and is attempting to find a final chunk to assemble the message before passing it on. It never sees a MSG though, as the client continues with HEL or OPN. It either appends this data to the previous message or - more likely to @Stasik0 - it disregards them, notices the error but does not reset the connection tracking, ignoring anything that is not a MSG/F.

Is there a reliable way to reproduce this behavior? Maybe we can add logging to the extend that receives from the tcp connection can be logged to diagnose where those packets are routed?

Stasik0, 2016-04-27:
Well, I got a virtualbox with win7 and a bridget network adapter. The win7 rund UaExpert 1.4.2, the server runs on linux host. With this constrlation, its pretty reproducible.

jpfr, 2016-06-06:
Can run run this in gdb, interrupt when the server hangs and give the backtrace?

Stasik0, 2016-06-09:
i cannot currently reproduce the hangs, but no communication can be established:
```
06/09/2016 12:48:45.800] info/network	TCP network layer listening on opc.tcp://xxx:16664
[06/09/2016 12:48:46.480] info/network	Connection 5 | New connection over TCP from 134.130.125.82:60538
[06/09/2016 12:48:46.481] info/channel	Connection 5 | SecureChannel 1 | OpenSecureChannel: Opened SecureChannel
[06/09/2016 12:48:46.481] debug/session	Try to use Session with token 483882979 but is not found
[06/09/2016 12:48:46.482] info/network	Decoding of message header failed on Connection 5
[06/09/2016 12:48:46.482] info/network	Connection 5 | Force closing the connection
[06/09/2016 12:48:46.482] info/network	Connection 5 | Connection closed from remote
[06/09/2016 12:48:51.656] info/network	Connection 5 | New connection over TCP from yyy:60539
[06/09/2016 12:48:51.657] info/channel	Connection 5 | SecureChannel 2 | OpenSecureChannel: Opened SecureChannel
[06/09/2016 12:48:51.662] info/network	Decoding of message header failed on Connection 5
[06/09/2016 12:48:51.662] info/network	Connection 5 | Force closing the connection
[06/09/2016 12:48:51.662] info/network	Connection 5 | Connection closed from remote
[06/09/2016 12:48:55.799] debug/channel	Connection 0 | SecureChannel 2 | SecureChannel has timed out
[06/09/2016 12:48:55.799] debug/channel	Connection 0 | SecureChannel 1 | SecureChannel has timed out
[06/09/2016 12:48:56.824] info/network	Connection 5 | New connection over TCP from yyy:60540
[06/09/2016 12:48:56.824] info/channel	Connection 5 | SecureChannel 3 | OpenSecureChannel: Opened SecureChannel
[06/09/2016 12:48:56.825] info/network	Decoding of message header failed on Connection 5
[06/09/2016 12:48:56.825] info/network	Connection 5 | Force closing the connection
[06/09/2016 12:48:56.825] info/network	Connection 5 | Connection closed from remote
[06/09/2016 12:49:01.996] info/network	Connection 5 | New connection over TCP from yyy:60541
[06/09/2016 12:49:01.996] info/channel	Connection 5 | SecureChannel 4 | OpenSecureChannel: Opened SecureChannel
[06/09/2016 12:49:01.997] info/network	Decoding of message header failed on Connection 5
[06/09/2016 12:49:01.997] info/network	Connection 5 | Force closing the connection
[06/09/2016 12:49:01.997] debug/network	Connection 5 | Received a MSG, but the connection is not established
[06/09/2016 12:49:01.997] info/network	Connection 5 | Connection closed from remote

```

xxx is my linux machine, yyy is a windows vm over a bridged network

jpfr, 2016-06-09:
Can you run tcpdump -i <interface> -w <outfile.pcap> on the linux machine?

> [06/09/2016 12:48:46.481] debug/session Try to use Session with token 483882979 but is not found

It is unusual that the clients tries to use a Session without creating or activating one.

Stasik0, 2016-06-15:
fixed by 331e2f7

closed 2016-06-15 by Stasik0