open62541/open62541 #598: Feature Lock/0.2 RC

ichrispa, 2016-04-28:
Hi everyone,

after a short discussion with @jpfr I would like to suggest that we feature lock the API and the current feature set for the 0.2 release. The ratio behind it: We have a lot of features that did not make it into the 0.1 branch. Among these are the namespace compiler, subscriptions, method calls and a ton of bug fixes. These features are actively used - at least by me - on a regular basis. I have not reverted to using 0.1 for ages. I also believe that most users are inclined to use the master branch more then the 0.1(.1) release. 

Several features are currently pending, such as asynchronous operations in the client, improvements to subscriptions, chunking and the event subsystem. The time frame for getting those features ready is uncertain (very uncertain on my side due to the current work load). If we wait for those, we might end up with "featuritis", stretching the 0.2 release point to months waiting for that 1 more cool feature to be implemented. Some features need a severe makeover, such as our take on threading (currently not well tested and only possible using liburcu).

My suggestion would thus be the following: 
1. Lock the current API and feature set in a release candidate. 
2. Attempt to get rid of the numerous pull requests and branches (which have grown immensly). 
3. Provide a 0.2 release
4. Any pending improvements to the 0.2 feature set, such as asynchronous subs in the client, get incorporated in 0.2.x when they become available.

New features and significant API changes will then target 0.3.

@jpfr also suggested to use master - which is by nature quite volatile - as a stablelish development branch and redirect users to use the latest release branch (0.2.x in this case) as default branch. I would second that, as we do not have a clean working scheme on where experimental features end up. The old development branch kind of vanished when subscritpions where merged into master.

Best regards,

Chris

uleon, 2016-04-28:
:+1: for 0.2. feature freeze and release

I'm not quite sure if I understand the branch scheme of @jpfr and @ichrispa correctly. We would expose an stable release branch for 0.2 that is maintainend in 0.2.x branches or tags and master would be pre-0.3 with continously experimental features ?

---+-- .... -- ... -- ... --- ..... master  
     \ 0.2 -- ...  -- 0.2.1 ...

Stasik0, 2016-04-29:
@uleon yes, it is actally the same scheme how we branched off 0.1 and 0.1.1

uleon, 2016-04-29:
:+1: for branching and advertising and maintaining 0.2 as stable build

markusgraube, 2016-05-01:
+1 for feature freezing and a stable 0.2 release

Furthermore I would like to suggest following a branching model which always have a stable master branch. For example like the following one: http://nvie.com/posts/a-successful-git-branching-model/

uleon, 2016-05-01:
This scheme is awesome. It would clearly help to raise the user experience of our stack.

jpfr, 2016-05-04:
I like the idea to have **proven default** that users can just take.

However, this scheme from the `nvie.com` post has one big disadvantage. Versions are just tags on the master branch. So it is difficult to make bugfix releases of old versions. For example releasing a 0.1.1 when 0.2 already exists.

Github allows us to define a **default branch that is not master**. For example, users who do a first clone will get branch 0.2 or whatever we define as default. Similarly, the default branch will show up on the github page first.

@markusgraube @uleon @ichrispa @Stasik0 
Would you be comfortable with the following?
- Development continues on master
- Nontrivial commits are only merged as pull requests when all tests are green
- We continue to make branches for *stable* releases 0.1, 0.2, ...
- The branch with the newest stable release (e.g. the upcoming 0.2) is marked as the default branch that users see without explictly switching
- Release branches may be updated for bugfix releases. Bugfix releases for old versions are decided by the core team

Stasik0, 2016-05-05:
Ist the default branch cloned when i execute "git clone"?

Stasik0, 2016-05-12:
Answering my own question, the *default* branch is the cloned when "git clone" is called, therefore +1 for @jpfr's idea, let us define the stable branch as default one and push patches to it. Furthermore we can use master branch as the bleeding-edge branch.

closed 2016-06-06 by jpfr