open62541/open62541 #599: Server Closes the connection after ~10 mins, (with UAExpert works fine)

BurhanHassan, 2016-04-29:
Hi,

I have noticed that when i connect my client to server, and let it remain idle, after ~10 mins the server closes the connection. How this time can be increased?

Regards,
Burhan

Stasik0, 2016-04-29:
the 10 minutes are in secureChannelLifeTime paramter exactly here: https://github.com/open62541/open62541/blob/5e87c1db6887d8782ac29360c30b1f0cf559ee20/plugins/ua_config_standard.c#L57

you can either increase this number, or just send some "keep-alive" requests to the server

BurhanHassan, 2016-04-29:
Hi Stasiko,
Thanks for the support.
I tried with first solution, doubled the value of secureChannelLifeTime  to 1200000, but still the connection is closed after 10 minute. 
Do this value have any upper limit?

Stasik0, 2016-04-29:
Actually the server decides how this value is set, you can capture some traffic and look the values in secure channel request and response.

So the best way would be to send some packets to the server once in 7-10 minutes.

closed 2016-05-04 by jpfr