open62541/open62541 #602: Running the OPC Foundation Conformance Testing Tools

jpfr, 2016-05-04:
Hey all,

Afaik everybody from the core team has access to the CTT (the current version is 1.2.336.256).

**Separate repository for the CTT config**
There now is a repository with a config for running the OPC Foundation conformance testing tools against our example server. Feel free to push your changes.
https://github.com/open62541/open62541-ctt

**Conformance Units that need to be all green**
With the upcoming 0.2 release, we should decide on the conformance units that need be all green.
Here's a list of the conformance units in the nano and micro profiles plus some things we have alrady implemented (node management, methods, ...). Some CU are unchecked because I have not tested them all.
Feel free to propose additional conformance units.

- [x] Address space base
- [x] Attribute Read (responsible: @jpfr)
- [ ] Attribute Write Index (responsible: @jpfr)
- [ ] Attribute Write Values (responsible: @jpfr)
- [ ] Base Info Core Structure
- [ ] Discovery Endpoints
- [ ] Discovery Find Servers self
- [ ] Method Call (responsible: @jpfr)
- [ ] Monitor Basic (@jpfr has a PR that fixes most issues)
- [ ] Monitor Items 2
- [ ] Monitor QueueSize 1
- [ ] Monitor Value Change
- [ ] Protocol TCP Binary
- [ ] Node Management Add Ref
- [ ] Node Management Delete Ref
- [ ] Node Management Add Node
- [ ] Node Management Delete Node
- [ ] Security None
- [ ] Security User Name Password
- [ ] Security None CreateSession ActivateSession
- [ ] Security Administration
- [ ] Session Minimum 1
- [ ] Session General Service Behaviour
- [ ] Session Base
- [ ] Subscription Basic (@jpfr has a PR that fixes most issues)
- [ ] Subscription Publish Discard Policy
- [ ] Subscription Minimum 1
- [ ] Subscription Publish Min 02
- [ ] View Basic
- [ ] View Minimum Continuation Point 01
- [ ] View RegisterNodes
- [ ] View TranslateBrowsePath (responsible: @jpfr)

**Responsible for the CU**
Can we assign a responsible to every conformance unit? Please come forward for the things you are interested in. I'll update the list in the first post accordingly.

ichrispa, 2016-06-02:
Hi @jpfr,

> Afaik everybody from the core team has access to the CTT (the current version is 1.2.336.256).

Not quite (and I would like to urge caution). Using and accessing CTT requires a cooperate membership in a certain foundation. Even if we had somehow gotten access to these tools, there might be a legal issue if we claim to use them.

> We aim for micro profile conformance

Though your efforts are admirable, I would like to point out that I do not aim for any CTT conformance. I need stack compatibility and a stable feature set. I'd rather have well written testsuits and a reliable API than a conformancy badge. 

As an added point: Introducing conformancy tests as a manditory 0.2 requirement is something I can definitely not handle on my time budget right now.

Non the less, a great effort on your part ;) thanks.

Best regards,

Chris

jpfr, 2016-06-03:
> Using and accessing CTT requires a cooperate membership in a certain foundation.

My organization is a member and we have an email "okay" to distribute the CTT in a limited scope for this project. I'll get to you outside this thread.

>  I would like to point out that I do not aim for any CTT conformance. I need stack compatibility and a stable feature set. I'd rather have well written testsuits and a reliable API than a conformancy badge.

Okay. But who do want to be compatible with exactly?
To me, that can only be **everybody else out in the world** aka **the spec**.
The public API for 0.2 should not change since the first release candidate.

The CTT are great. They are like a large set of unit/integration tests we don't have to write ourselves.
If we had a server to spare, we could even set up continuous integration for them.

> As an added point: Introducing conformancy tests as a manditory 0.2 requirement is something I can definitely not handle on my time budget right now.

I don't want to add any tests. That's only a measure towards the real goal.
The goal is: No known defects and non-compatibilities for the implemented features in the release.
At this stage we need to start building trust. So yes, **the release flies only once we pass the CTT**. But I see no major blockers. A couple more weeks of part-time issue solving and we should be fine.

At this point I'd also like to thank @Stasik0. I think he ticked off the most boxes in the list so far!

Stasik0, 2016-07-28:
@jpfr I reverted a test-related commit of yours - be aware 00e7e8acc5818d7f15d42cba161fc3c014ed1aff

jpfr, 2016-09-18:
all done

closed 2016-09-18 by jpfr

masongk, 2018-12-13:
Hi @jpfr 

I would like to run CTT on open62541 stack. I could not find the project repository for CTT configuration ([https://github.com/open62541/open62541-ctt](https://github.com/open62541/open62541-ctt)). Could please share the latest configuration or point me to the repository (open62541-ctt)

Thanks

jpfr, 2018-12-13:
We were made aware that the CTT project configuration contains a copy of the javascript test definitions.
So parts of our CTT project configuration may fall under copyright.
We removed the repository.

If you are a Foundation member with access to the CTT, I can make the project configuration available privately. Please send a mail.