open62541/open62541 #604: networklayer_tcp fixing select()

NetoLe, 2016-05-10:
First: I'm not into git. I tried your guide from http://blog.ericbmerritt.com/2011/09/21/commit-hygiene-and-git.html but it didn't work because I have no permission (so it tells me) for commiting anything. I'm feeling a bit dumb.

I wanted to fix a small problem in networklayer_tcp.c.
This would be the description:

> 
> fixed socket-overflow on some devices
> 
> This is made regarding an embedded device with proprietary networklayer-functions. It is geared to socket-functions from the Unix-man page of select(), which says that it only counts the events which are watched. If no error-event is watched it won't be counted, which leads to open sockets even if they should be closed because they were shutdown'd before, because the select isn't triggered because it doesn't watch for errors, when the parameter for the error-set is NULL. It now works fine and closes the specific sockets as it should. Sureley this isn't a noticable problem on big desktop-machines with a lot of sockets but if you are on an embedded device reduced to 30 or something you will experience a run out of free sockets very fast after some connects and disconnects. This fixes prior problems regarding conflicts for those who had problems with UaExpert. This relates to https://github.com/open62541/open62541/issues/576 and https://github.com/open62541/open62541/issues/517

and this is the c-file as a txt (just change the ending): 
[networklayer_tcp.txt](https://github.com/open62541/open62541/files/256809/networklayer_tcp.txt)

jpfr, 2016-05-10:
Thanks for the patch!

Just for completeness, here is the diff.
````diff
394c394
<     fd_set fdset;
---
>     fd_set fdset, errset;
395a396,399
> <<<<<<< HEAD
>     setFDSet(layer, &errset);
>     struct timeval tmptv = {0, timeout};
> =======
396a401
> >>>>>>> origin/master
398c403
<     resultsize = select(highestfd+1, &fdset, NULL, NULL, &tmptv);
---
>     resultsize = select(highestfd+1, &fdset, NULL, &errset, &tmptv);
429,430c434,437
<         if(!UA_fd_isset(layer->mappings[i].sockfd, &fdset))
<             continue;
---
>         if(UA_fd_isset(layer->mappings[i].sockfd, &errset)) {}
>         else if(!UA_fd_isset(layer->mappings[i].sockfd, &fdset)) {
>           continue;
>         }
634a642
>
```
It looks like we can just merge this patch.
I'd just add a pull request to see if all platforms compile fine and run all tests.

NetoLe, 2016-05-10:
I noticed that there was a change I didn't involve:

`struct timeval tmptv = {0, timeout};`
should be:
`struct timeval tmptv = {0, timeout * 1000};`
due to https://github.com/open62541/open62541/commit/21e81840c218c343eef2af61476135168a600839

NetoLe, 2016-05-10:
ok I think I figured out now how git works ;) I created a pull request with the changes.

closed 2016-05-10 by NetoLe