open62541/open62541 #607: Client closes connection to asynchronous Subscriptions after a short time (~2 min).

NetoLe, 2016-05-13:
I experienced a small problem.

I'm using my old client (with the current open62541-Stack (master)), which has an implemented
UA_Client_Subscriptions_manuallySendPublishRequest

that is called every 500 ms.
So the values in the device aren't updated every 500 ms, let's say it are 2 Seconds.

What I experienced is, that it works just for a short time. Every request gets its response immediately. But after ~ 2 min the server doesn't answer the request and the client starts closing the connection. 

A Wireshark trace (with the middle messages cut out) is here:
[client_ending.zip](https://github.com/open62541/open62541/files/263249/client_ending.zip)

With UaExpert everything works fine for a long time.

jpfr, 2016-05-13:
From what I see in the dump, the client sends a MSGF message where it is not supposed to.
Also the content of the message marked TCP segment of a reassembled PDU is not what I would expect.

[image]

Can you run your client with UA_LOGLEVEL set to 100 or less?
That gives additional output that might hint at the reasons for that behaviour.

Further, I just fixed a bug in the acknowledgement of publish responses. You might reduce the network traffic by using the latest master for the client.

NetoLe, 2016-05-13:
hm, so I've written an MFC-Application for the client, those printf's are not going anywhere then. 

But I think the mistake is somewhere on my side. My Debugging-Client with the same stack-version works forever without cutting the connection. How much does the stack rely on ERRNO? I once had that problem that a sprintf I used triggered the ERRNO which caused the Client to stop working.

NetoLe, 2016-05-13:
oh, I noticed something. In the wireshark-Trace "Array of simple type" is constantly increasing and is filled with 0x807a0000. Using my other client it's always of size -1.

What is this array exactly for?

NetoLe, 2016-05-13:
ok, I tried something different. When I set my cycle to 5000 ms, so 5s then the array doesn't increase. When i set it to 2s it's the constant size of 2 with one result beeing 0x807a0000.

Is there a related point between the number of monitored items and the time in which they are requested? as you see I have many monitoredItems (~50).

After that I tried to recreate this with UaExpert monitoring a lot of nodes but this doesn't work.

My intention is that the problem is caused by the short time between request and response and that there an error with the sequence numbers occurs. And I don't get it why the server responds immediately to a manual request that is called cyclic and with a delay (till the next change) when monitoring with UaExpert.

ichrispa, 2016-05-20:
@NetoLe: I have a similar use-case to your (old client, new server) that heavily relies on subscriptions. I cannot see the reconnect problem on my side.

Please make sure the client sends RenewSecureChannel or Publishrequests at regular intervals to keep the connection alive.

jpfr, 2016-06-07:
> Is there a related point between the number of monitored items and the time in which they are requested? as you see I have many monitoredItems (~50).

No. Up to the point where you have enough computational resources.
The sampling interval is set individually for every MonitoredItem.
And the publication interval is set for the subscription where the MoinitoredItem is attached.

> And I don't get it why the server responds immediately to a manual request that is called cyclic and with a delay (till the next change) when monitoring with UaExpert.

Our client is not yet async. So we set the subscription settings to `maxKeepAliveCount=1`. Then the server responds to every publish request and does not put responses "on hold" until a subscription wants to publish.

closed 2016-06-07 by jpfr