open62541/open62541 #614: Client-Subscription: First callback from a float node has type double

Root-Core, 2016-05-19:
Hello,

I am using the open62541 server, i add some nodes as MonitoredItems to a subscription.
One is a float typed node (DataSource).

The first call of "UA_Client_Subscriptions_manuallySendPublishRequest" gives me the correct values (eg. Int32), but the float node gives me a DataValue of the type double (Value: 0.0).

Following calls work fine. (Type float and correct values)

I will try to investigate this issue, but i am short on time right now.

Regards, Root-Core.

jpfr, 2016-06-06:
What is a float node in this context?

Root-Core, 2016-06-06:
Eg. the CPU temp of the device, which is a DataSource-node, server side.
I am using requests right now, so i can't tell if this issue is still present.

jpfr, 2016-06-07:
> The first call of "UA_Client_Subscriptions_manuallySendPublishRequest" gives me the correct values (eg. Int32), but the float node gives me a DataValue of the type double (Value: 0.0).

So you have int32 variables that give the correct value and a double variable that first gives 0.0 and then switches to the correct value?

Root-Core, 2016-06-07:
A Int32 variable gives me the correct value with the correct type, but a float variable gives me a response with a double typed value, which is always 0.0..

The second iteration delivers the correct type and the correct value.

jpfr, 2016-06-07:
Int and double are handled exactly the same internally.
So this is an unlikely bug.

Are you using data sources or "normal" variables?
Are you sure that the double is !=0 before the subscription is started?

Root-Core, 2016-06-07:
The float equals to the CPU temp and is mirrored in UA Expert, so it is not 0.
I am using DataSources.

To conclude:
The server holds a float value from a DataSource, which is valid.
The client gets a double value (always 0.0) at the first iteration of UA_Client_Subscriptions_manuallySendPublishRequest and the correct value at following iterations.

As i said, I'm working on other stuff right now, therefore I have not checked, if this behavior is still present.

jpfr, 2016-06-30:
Hey @Root-Core,

I'm not able to reproduce this. Can you provide a Wireshark dump or similar?

Pro, 2017-09-14:
@Root-Core is this still an open issue?

Root-Core, 2018-01-16:
I actually can't tell if there is still an issue here, as I had to use readRequests. But I have to reactivate the subscriptions in our project and will have a look into it. This shouldn't stop the 0.3 release, if you can't reproduce.

closed 2018-01-17 by Pro