open62541/open62541 #617: UA_Client_connect - getendpoint problem - sdk v0.2

ilike28, 2016-05-24:
I use client in the ubuntu of virtual box and open a opc foundation server on the win 7.

The server address is opc.tcp://pc20140604c:62541/Quickstarts/ReferenceServer.
I can ping pc20140604c in the ubuntu .
But ,when I use UA_Client_connect function which  show the "GetEndpointRequest fail" durning the connect state !

Does anyone know how to fix this problem??

Thanks~~

[image]

Stasik0, 2016-05-27:
hey, could you post a link to the server you are using?

jpfr, 2016-05-28:
I just pushed a patch to 0.2 so that more statuscodes are logged for the client.

@ilike28 Can you just redo your tests with the current 0.2 branch and post the logs?
My guess is that the server allows encrypted connections only.

ilike28, 2016-05-30:
I got the following statecode by the new v0.2 SDK 

[image]

Stasik0, 2016-05-30:
@ilike28 it seems to be our bug, getting endpoints should be possible without encryption, i will look into it

Stasik0, 2016-05-30:
@ilike28 please try it now, we had a problem with the local config missmatch

ilike28, 2016-05-31:
It can work now !!
Thank you very~~~~~ much ^^!

[image]

closed 2016-05-31 by ilike28