open62541/open62541 #618: Problems with macros in queue.h

AugustinusVonHippo, 2016-05-25:
Hello, 

I want to use the RC 0.2 code of `open62541.h` and `open62541.c` to set up an OPC UA server. 
I use gcc 4.5.2 and wcripts to build a target for an embedded system.

I get some compiler errors that I cannot resolve myself, for example:
```
...\open62541.c:2086:5: error: expected specifier-qualifier-list before 'SIMPLEQ_HEAD'
...\open62541.c:12406:5: warning: implicit declaration of function 'LIST_FOREACH_SAFE'
```
Stackoverflow and other sources stated that the first error occurs when a qualifier is used, that is not specified before. In this forum was an issue with this topic before, where the problem was solved by changing the order of the amalgamated files. 
But I think that this is not a matter of sequencial order. All those places, where the macros of `queue.h` are used are after the amalgamated file `queue.h`. 
If I replace the call of the macro with the code of the macro (with appropriate replacement of variables), it works. But I don't want to replace nearly 200 lines of code...

Do you have any suggestions what I can do to resolve this issue?
Regards,
Augustinus

jpfr, 2016-05-25:
Can you show us the call to gcc?

`gcc-4.6 -std=c99 open62541.c server.c -o server`
That works fine for me. Do you need gcc 4.5.2 or can you also try with a different version?

I'm a bit surprised since we use different gcc versions for continuous integration.
And we haven't seen that error there.

AugustinusVonHippo, 2016-05-27:
Hello,

that's our gcc call: 
```
`\arm-none-eabi-gcc.exe', '-ffunction-sections', '-fdata-sections', '-msoft-float', '-mfpu=vfp', '-mfloat-abi=soft', '-Wall', '-Wredundant-decls', '-Wno-inline', '-Winit-self', '-mlong-calls', '-mapcs', '-mthumb-interwork', '-fshort-enums', '-fno-common', '-mcpu=arm966e-s', '-std=c99', '-O0', '-g', '-gdwarf-2', '-mthumb', '-IC:\\...\\Includes', (very much includes with '-IC') '-DTARGET_MAJOR=2', (a lot of other -D variables) '..\\Led.c', '-c', '-o', '..\\Led.o'
```

Is there something special?
Thanks.

Stasik0, 2016-05-27:
hey @AugustinusVonHippo, well the last bug ended up without a solutoin. I do not see any specialities in your gcc call? What you can try:
* use another i.e. newer queue.h version e.g. this one https://ftp.netbsd.org/pub/NetBSD/NetBSD-current/src/sys/sys/queue.h
* execute a preompiler run for the header with a newer e.g. 4.6 gcc or just replace the one macro with a simple script

hth

jpfr, 2016-05-28:
Hey @AugustinusVonHippo ,

I had a closer look at arm-non-eabi cross-compilers for gcc 4.6.
Specifically this version: https://launchpad.net/gcc-arm-embedded/+milestone/4.6-2012-q4-update

The cross-compiler comes with a custom file `/arm-none-eabi/include/sys/queue.h`.
However, the version they distribute differs from ours (/deps/queue.h, taken from OpenBSD).
Some features are missing. Notably the `SIMPLEQ` and all the `SAFE` iterators.

My guess is: Your code is loading the gcc-queue.h with another dependency before our header. Then, the include guards
```c
#ifndef _SYS_QUEUE_H_
#define	_SYS_QUEUE_H_
```
prevent our queue.h to be evaluated.

Can you run gcc with the `-H` or `-M` option to show the hierarchy of includes?
https://gcc.gnu.org/onlinedocs/gcc/Preprocessor-Options.html#Preprocessor-Options

AugustinusVonHippo, 2016-05-30:
Hello @jpfr
 
thank you very much for your saturday evening work! Your guess was right. 
I changed the file the compiler uses into the OpenBSD queue.h and it works now. This is probably not the best solution. Is there another way to avoid this interference?

Thank you,
Augustinus

jpfr, 2016-05-31:
Well, if your compiler pulls in queue.h (which it shouldn't, imho) with the standard libraries (which it shouldn't, imho), then you can do three things:
- exchange the default queue.h (which you did)
- find out at which point the include happens (remember the -H and -M options to find that out) and include the OpenBSD queue.h before that so you get the include guard.
- switch to another tool chain

Since ARM is an important target, I'd like to fix this once and for all.
Take open62541.c and move the line `#include "open62541.h"` down below the following queue.h. Does that help?

AugustinusVonHippo, 2016-05-31:
Hello @jpfr,

thank you for outlining the options. For the moment I'll stay with the first solution. Maybe I do have to add only the missing makros in a further step, but I don't hope so.
I did your task at the end of your answer, but &mdash; it doesn't help anything. Why should it?

Regards,
Augustinus

jpfr, 2016-05-31:
> I did your task at the end of your answer, but — it doesn't help anything. Why should it?

It might be that the wrong queue.h was included via open62541.h.
But you would only seed the difference if you still had the old queue.h bundled with arm-non-eabi-gcc.

I close this issue then...

closed 2016-05-31 by jpfr