open62541/open62541 #620: UA_Client  - UACTT -Server test -initializes.js problem - sdk v0.2

ilike28, 2016-05-31:
The UACTT Server test run the initializes.js  before all of the test case.
And it got the following issue.
Read.Response.ResponseHeader.ServiceResult is Bad: BadInternalError(0x80020000)

[image]

ilike28, 2016-06-01:
update the fail point @@~
Does anyone get the same issue during CTT Server test?

[image]

[image]

Stasik0, 2016-06-01:
@ilike28 this is what i see here with the latest commit c1d7abf7de77bdd5928ea846513ef45058f8bf72

[image]

Which CTT version are you using?

ilike28, 2016-06-01:
I use version 1.2.336.256.
And download the CTT test Project from : https://github.com/open62541/open62541-ctt.
@@

ilike28, 2016-06-02:
I upload a video to show my building and testing step~ 
https://drive.google.com/file/d/0B5VqbFqMszDhYzljcnFITktuRDg/view?usp=sharing

jpfr, 2016-06-03:
Fixed in commit 6d65901096afc79b9e7ff1835c2a4a95481638e0.
The generated namespace contains variants that are never given any content.
Those are handled correctly now.

closed 2016-06-03 by jpfr

ilike28, 2016-06-04:
@jpfr 
Thanks ^^