open62541/open62541 #621: Server Building with XML problem

ilike28, 2016-05-31:
I use the following command.
 sudo cmake -DUA_BUILD_EXAMPLECLIENT=On -DUA_BUILD_EXAMPLESERVER=On -DUA_ENABLE_SUBSCRIPTIONS=ON -DUA_BUILD_EXAMPLE=ON -DUA_ENABLE_GENERATE_NAMESPACE0=ON -DUA_ENABLE_EXTERNAL_NAMESPACES=ON -DUA_ENABLE_GENERATE_NAMESPACE0_FILE=ON

and use the default Opc.Ua.NodeSet2.xml  ,then got the following error.  @@

[image]

[image]

ichrispa, 2016-06-02:
Hi @ilike28 

would you mind posting the output as text instead of cut-off screenshots?

Also: Why would you be building as root!?

ichrispa

ilike28, 2016-06-03:
@ichrispa 

Hi~~~

I use ubuntu 16.04  64Bit on virtualbox.

=====================================================================

opcua@opcua-VirtualBox:~$ gcc --version
gcc (Ubuntu 5.3.1-14ubuntu2.1) 5.3.1 20160413
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

=====================================================================

opcua@opcua-VirtualBox:~/aaa/open62541-0.2/build$ cmake -DUA_BUILD_EXAMPLECLIENT=On -DUA_BUILD_EXAMPLESERVER=On -DUA_ENABLE_SUBSCRIPTIONS=ON -DUA_BUILD_EXAMPLE=ON -DUA_ENABLE_GENERATE_NAMESPACE0=ON -DUA_ENABLE_EXTERNAL_NAMESPACES=ON -DUA_ENABLE_GENERATE_NAMESPACE0_FILE=ON ..
fatal: Not a git repository (or any of the parent directories): .git
-- Git failed (not a repo, or no tags). Build will not contain git revision info.
-- Git version: unknown--git-commit-id-unknown
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    UA_ENABLE_GENERATE_NAMESPACE0_FILE

-- Build files have been written to: /home/opcua/aaa/open62541-0.2/build
opcua@opcua-VirtualBox:~/aaa/open62541-0.2/build$ make
Scanning dependencies of target open62541-object
[  2%] Building C object CMakeFiles/open62541-object.dir/src/ua_types.c.o
[  4%] Building C object CMakeFiles/open62541-object.dir/src/ua_types_encoding_binary.c.o
[  6%] Building C object CMakeFiles/open62541-object.dir/src_generated/ua_types_generated.c.o
[  8%] Building C object CMakeFiles/open62541-object.dir/src_generated/ua_transport_generated.c.o
[ 10%] Building C object CMakeFiles/open62541-object.dir/src/ua_connection.c.o
[ 12%] Building C object CMakeFiles/open62541-object.dir/src/ua_securechannel.c.o
[ 14%] Building C object CMakeFiles/open62541-object.dir/src/ua_session.c.o
[ 17%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server.c.o
/home/opcua/aaa/open62541-0.2/src/server/ua_server.c: In function ‘UA_Server_addExternalNamespace’:
/home/opcua/aaa/open62541-0.2/src/server/ua_server.c:75:19: error: conversion to ‘UA_UInt32 {aka unsigned int}’ from ‘size_t {aka long unsigned int}’ may alter its value [-Werror=conversion]
  UA_UInt32 size = server->externalNamespacesSize;
                   ^
/home/opcua/aaa/open62541-0.2/src/server/ua_server.c:79:43: error: conversion to ‘UA_UInt16 {aka short unsigned int}’ from ‘size_t {aka long unsigned int}’ may alter its value [-Werror=conversion]
  server->externalNamespaces[size].index = server->namespacesSize;
                                           ^
/home/opcua/aaa/open62541-0.2/src/server/ua_server.c:80:29: error: conversion to ‘UA_UInt16 {aka short unsigned int}’ from ‘size_t {aka long unsigned int}’ may alter its value [-Werror=conversion]
  *assignedNamespaceIndex  = server->namespacesSize;
                             ^
/home/opcua/aaa/open62541-0.2/src/server/ua_server.c:83:33: error: passing argument 2 of ‘UA_Server_addNamespace’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  UA_Server_addNamespace(server, url);
                                 ^
In file included from /home/opcua/aaa/open62541-0.2/src/server/ua_server_internal.h:5:0,
                 from /home/opcua/aaa/open62541-0.2/src/server/ua_server.c:2:
/home/opcua/aaa/open62541-0.2/include/ua_server.h:189:21: note: expected ‘const char *’ but argument is of type ‘const UA_String * {aka const struct <anonymous> *}’
 UA_UInt16 UA_EXPORT UA_Server_addNamespace(UA_Server *server, const char* name)
                     ^
cc1: all warnings being treated as errors
CMakeFiles/open62541-object.dir/build.make:274: recipe for target 'CMakeFiles/open62541-object.dir/src/server/ua_server.c.o' failed
make[2]: *** [CMakeFiles/open62541-object.dir/src/server/ua_server.c.o] Error 1
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/open62541-object.dir/all' failed
make[1]: *** [CMakeFiles/open62541-object.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
opcua@opcua-VirtualBox:~/aaa/open62541-0.2/build$

jpfr, 2016-06-04:
You don't need the -DUA_ENABLE_EXTERNAL_NAMESPACES=ON option to generate the namespace from XML.
Anyway, fixed in c4b386c4992ca498ee70765720442d7eff892910.

closed 2016-06-04 by jpfr