open62541/open62541 #625: UA_WriteRequest_deleteMembers corrupts memory

Root-Core, 2016-06-03:
If the nodeId of one nodesToWrite is an stored UA_NODEID_STRING_ALLOC, the client crashes.

I get this NodeID from a class, which stores an UA_NodeId:
`const UA_NodeId GetNodeId() { return nodeId_; }`

I know, we alloc memory in order to store the string, so we are free'ing where the pointer points to on deleteMembers.. but this should not happen here, i guess.

So we should discuss, if this is a valid behavior or a bug.
As a workaround, i now copy the whole UA_NodeId object.

Thank you, Root-Core.

jpfr, 2016-06-03:
The behaviour you describe is intended.
Every malloced content of the WriteRequest is freed by deleteMembers.

So yes, copy the node with UA_NodeId_copy if you don't want to loose the original after 
UA_WriteRequest_deleteMembers.

Or, if you want to be sneaky, run UA_NodeId_init on the nodeid inside the WriteRequest prior to calling UA_WriteRequest_deleteMembers to prevent the string from being freed.

Can we close?

Root-Core, 2016-06-06:
Okay, that the deleteMembers function is intended to free everything malloced "to the Object" was clear to me, but i did not assume, that this should happen to something like the origin of the "copied" NodeId.

I thought, as the NodeId is not specificly malloced to the WriteRequest, it may shouldn't get freed either. But as you say it is intended, it should work for me.

Thanks for the idea with the init, i'll test that out.
Regards, Root-Core.

jpfr, 2016-06-06:
The deleteMembers function only sees the pointer. It has no way to figure out where the pointer comes from. That's the downside from using a language without garbage collection.

closed 2016-06-06 by jpfr