open62541/open62541 #628: Tutorial doenst work

SvenManier, 2016-06-07:
I tried the tutorial: http://open62541.org/doc/current/tutorial_server_firstSteps.html
but I get these messages:
```
[smanier@wlan10-9-49-119 myApp]$ gcc -Wl,-rpath,`pwd` -I ./include -L ./ ./myServer.c -o myServer  -lopen62541
./myServer.c: In Funktion »main«:
./myServer.c:11:28: Fehler: »UA_ServerConfig_standard« nicht deklariert (erste Benutzung in dieser Funktion)
   UA_ServerConfig config = UA_ServerConfig_standard;
                            ^
./myServer.c:11:28: Anmerkung: jeder nicht deklarierte Bezeichner wird nur einmal für jede Funktion, in der er vorkommt, gemeldet
./myServer.c:12:30: Fehler: zu viele Argumente für Funktion »UA_ServerNetworkLayerTCP«
   UA_ServerNetworkLayer nl = UA_ServerNetworkLayerTCP(UA_ConnectionConfig_standard, 16664, logger);
                              ^
In file included from ./myServer.c:6:0:
./include/networklayer_tcp.h:17:1: Anmerkung: hier deklariert
 UA_ServerNetworkLayerTCP(UA_ConnectionConfig conf, UA_UInt16 port);
 ^
```

jpfr, 2016-06-07:
The examples are compiled by travis after every commit. So they do work for sure.

Which version of open62541 do you use?
The "current" documentation refers to the current master and 0.2 branch.
Just download the v0.2-rc1 release candidate.

SvenManier, 2016-06-07:
Thanks for the fast reply =)
I downloaded the latest 0.2 Branch again and did the Tutorial, still the same
```
[smanier@wlan10-9-49-85 Arbeit]$ mkdir myApp
[smanier@wlan10-9-49-85 Arbeit]$ cd myApp/
[smanier@wlan10-9-49-85 myApp]$ mkdir include
[smanier@wlan10-9-49-85 myApp]$ cp ../open62541-0.2/include/* ./include/
[smanier@wlan10-9-49-85 myApp]$ cp ../open62541-0.2/plugins/*.h ./include/
[smanier@wlan10-9-49-85 myApp]$ cp ../open62541-0.2/build/src_generated/*.h ./include/
[smanier@wlan10-9-49-85 myApp]$ cp ../open62541-0.2/build/*.so .
[smanier@wlan10-9-49-85 myApp]$ tree
.
├── include
│   ├── logger_stdout.h
│   ├── networklayer_tcp.h
│   ├── networklayer_udp.h
│   ├── ua_client.h
│   ├── ua_client_highlevel.h
│   ├── ua_config.h
│   ├── ua_config.h.in
│   ├── ua_config_standard.h
│   ├── ua_connection.h
│   ├── ua_constants.h
│   ├── ua_job.h
│   ├── ua_log.h
│   ├── ua_nodeids.h
│   ├── ua_server_external_ns.h
│   ├── ua_server.h
│   ├── ua_transport_generated_encoding_binary.h
│   ├── ua_transport_generated.h
│   ├── ua_types_generated_encoding_binary.h
│   ├── ua_types_generated.h
│   └── ua_types.h
└── libopen62541.so

1 directory, 21 files
[smanier@wlan10-9-49-85 myApp]$ ls
include  libopen62541.so
[smanier@wlan10-9-49-85 myApp]$ nano myServer.c
[smanier@wlan10-9-49-85 myApp]$ gcc -Wl,-rpath,`pwd` -I ./include -L ./ ./myServer.c -o myServer  -lopen62541
./myServer.c: In Funktion »main«:
./myServer.c:11:28: Fehler: »UA_ServerConfig_standard« nicht deklariert (erste Benutzung in dieser Funktion)
   UA_ServerConfig config = UA_ServerConfig_standard;
                            ^
./myServer.c:11:28: Anmerkung: jeder nicht deklarierte Bezeichner wird nur einmal für jede Funktion, in der er vorkommt, gemeldet
./myServer.c:12:30: Fehler: zu viele Argumente für Funktion »UA_ServerNetworkLayerTCP«
   UA_ServerNetworkLayer nl = UA_ServerNetworkLayerTCP(UA_ConnectionConfig_standard, 16664, logger);
                              ^
In file included from ./myServer.c:6:0:
./include/networklayer_tcp.h:17:1: Anmerkung: hier deklariert
 UA_ServerNetworkLayerTCP(UA_ConnectionConfig conf, UA_UInt16 port);
 ^
```

closed 2016-06-07 by jpfr

jpfr, 2016-06-07:
The fix is pushed.

Alternatively, you can download the open62541.c/h from http://open62541.org and build against that.

commit 326e99d6 (2016-06-07): fix #628; an issue in the firststeps tutorial