open62541/open62541 #629: Nodes of a type with properties, crashes UA Expert

Root-Core, 2016-06-07:
Hello!

I am adding a DataSource of the AnalogItemType, as such it has the properties "EURange", "EngineeringUnits" and "InstrumentRange".

Formerly i added them by hand, but there seems to be a new behavior, that also creates this subnodes.
If you try to read them, UA Expert will crash. Furthermore, this subnodes are not recognized as properties, they seem to be a VariableNode.

I am pretty sure, that UA Expert crashes, as the value of the nodes isn't set, this also happens to other nodes, if you don't set the value.

Is there a way to suppress this automatic generation of subnodes?

Thanks and regards, Root-Core.

jpfr, 2016-06-07:
Hello @Root-Core ,

the automatic generation of subnodes will happen during the instantiation of a variable.
They are probably required for the variabletype you are using.

The crashing of UAExpert seems to have started with this commit:
https://github.com/open62541/open62541/commit/6d65901096afc79b9e7ff1835c2a4a95481638e0
It implements empty variants containing no value at all.

Null variants are mandated by the standard:
> Variants can be empty. An empty Variant is described as having a null value and should be
treated like a null column in a SQL database. A null value in a Variant may not be the same as
a null value for data types that support nulls such as Strings. Some Development Platforms
may not be able to preserve the distinction between a null for a DataType and a null for a
Variant. Therefore Applications shall not rely on this distinction.

I will open a thread on the UnifiedAutomation forums.

Root-Core, 2016-06-07:
Hello @jpfr ,

the problem here is, that they are not referenced as properties. And some of the properties are not mandatory, so i do not want to add them either.

In the past, there was no automatic generation for me.
How should I modify a generated node, eg. to make it a property.

Okay, thank you for the clarification about UAExpert.

jpfr, 2016-06-15:
Unified Automation has fixed the crashing behaviour of UAExpert.
The fix will be contained in the next release.

Further, we adjusted the behavior from our side. So you can just use the current 0.2 branch.

closed 2016-06-15 by jpfr