open62541/open62541 #630: Client can't read structur type

Root-Core, 2016-06-08:
Hello,

I try to read an structure (UA_Range) from an server.
This is not possible, as i get a BadInternalError or a BadDecodingError status code.

I tried it with a call of UA_Client_readValueAttribute and with a handcrafted function.
Both are capable of reading standard types, like float.

Is this a known issue / not implemented yet?

Thank you, Root-Core.

jpfr, 2016-06-08:
In order to keep the binaries small, we only generate a subset of all datatypes.
https://github.com/open62541/open62541/blob/0.2/tools/schema/datatypes_minimal.txt

We should probably make this optional and generate all types per default.
In the meantime, you can just add Range to datatypes_minimal.txt.
The error-code is indicating at a bug. Normally, the decode-functions should return a variant with a binary blob if the type was not recognized.

Root-Core, 2016-06-08:
As I added it to the datatypes_minimal.txt (the server and client are running on the same codebase, on the same device), so i think there is something broken.

jpfr, 2016-06-08:
Fixed on the branches master and 0.2

closed 2016-06-08 by jpfr

Root-Core, 2016-06-09:
Hello @jpfr, thanks a lot for the fix!

I actually do get an ExtensionObject, which is better than nothing, I guess. 
But shouldn't it be decoded as an UA_Range object, if this DataType is known?

Root-Core, 2016-06-09:
Okay, sorry.. an UA_Range actually is an ExtensionObject.

I got into a mistake because you said "the decode-functions should return a variant with a binary blob if the type was not recognized.", my fault.

Thanks for your time.

jpfr, 2016-06-09:
Good point. You get an UA_Range and not an UA_Range inside an ExtensionObject now.

Root-Core, 2016-06-09:
Thank you very much! This seems to be more intuitive. I will test it later. ;)

Root-Core, 2016-06-09:
Works like a charm! Thank you.

commit 5c321214 (2016-06-08): fix #630
commit 09db9014 (2016-06-08): fix #630
commit 1901dc1c (2016-06-09): return the type and not an extensionobject if possible; fix #630 2nd time