open62541/open62541 #632: The first session GUID is always the same

Root-Core, 2016-06-09:
Hello,

i saw, that the logging shows me always the same GUID (152CA78D-6003-F3BF-BB7BEEFEEFBE) for the first session.
This issue is tested on and valid for arm-linux and Windows_x86.

It also seems, that the server sometimes sets this same GUID for different sessions, which leads the server to confuse the sessions and get nonfunctional (you can not connect). But I need to check this again.

Regards, Root-Core.

jpfr, 2016-06-09:
The session guids are generated randomly.
However, the rng seed is initialized deterministically here:
https://github.com/open62541/open62541/blob/0.2/src/ua_types.c#L19

The internal rng should not be used for cryptographically relevant entropy.
And yes, the first session will have the same name.

However, it is very very unlikely that the same id is generated twice during a single run of the server.
Did that happen?

Root-Core, 2016-06-09:
I investigate this right now.. here is some logging.
There are different IPs, but equal IDs on different SecureChannels.

```
Connection 41 | New connection over TCP from 192.168.250.72:46214
Connection 41 | SecureChannel 1 | OpenSecureChannel: Opened SecureChannel
Connection 17 | New connection over TCP from 192.168.250.46:52504
Connection 41 | SecureChannel 1 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | ActivateSession: Session activated
Connection 17 | SecureChannel 2 | OpenSecureChannel: Opened SecureChannel
Connection 41 | SecureChannel 1 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | ActivateSession: Detach from old channel
Connection 17 | SecureChannel 2 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | ActivateSession: Session activated
Connection 41 | SecureChannel 1 | CloseSecureChannel
Connection 41 | Connection closed from remote
Connection 18 | New connection over TCP from 192.168.250.72:46215
Connection 18 | SecureChannel 3 | OpenSecureChannel: Opened SecureChannel
Connection 17 | SecureChannel 2 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | ActivateSession: Detach from old channel
Connection 18 | SecureChannel 3 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | ActivateSession: Session activated
Connection 17 | SecureChannel 2 | CloseSecureChannel
Connection 17 | Connection closed from remote
Connection 17 | New connection over TCP from 192.168.250.46:52512
Connection 17 | SecureChannel 4 | OpenSecureChannel: Opened SecureChannel
Connection 18 | SecureChannel 3 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | ActivateSession: Detach from old channel
Connection 17 | SecureChannel 4 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | ActivateSession: Session activated
Connection 18 | SecureChannel 3 | CloseSecureChannel
Connection 18 | Connection closed from remote
Connection 18 | New connection over TCP from 192.168.250.72:46218
Connection 18 | SecureChannel 5 | OpenSecureChannel: Opened SecureChannel
Connection 17 | SecureChannel 4 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | ActivateSession: Detach from old channel
Connection 18 | SecureChannel 5 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | ActivateSession: Session activated
Connection 17 | SecureChannel 4 | CloseSecureChannel
[...]
Connection 43 | SecureChannel 29 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | ActivateSession: Detach from old channel
Connection 17 | SecureChannel 30 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | ActivateSession: Session activated
Connection 43 | SecureChannel 29 | CloseSecureChannel
Connection 43 | Connection closed from remote
Connection 43 | New connection over TCP from 192.168.250.72:46274
Connection 43 | SecureChannel 31 | OpenSecureChannel: Opened SecureChannel
Connection 17 | SecureChannel 30 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | ActivateSession: Detach from old channel
Connection 43 | SecureChannel 31 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | ActivateSession: Session activated
Connection 17 | SecureChannel 30 | CloseSecureChannel
Connection 17 | Connection closed from remote
```

Stasik0, 2016-06-09:
as far as I see the session is activated only once "ActivateSession: Session activated" and then gets detached and re-attached to another secure-channel? What client or client setting are you using?

Root-Core, 2016-06-09:
The session is actually activated, after it got "Detach(ed) from old channel", or is this just a misleading logging?

These are not the same clients reconnecting, therefore they should not get the same session id, if i understood it correctly.

I am using the open62541 client (2.0 branch), UAExpert 1.4.2 (256) and UAExpert 1.3.8 (157).. all with default config.

Stasik0, 2016-06-09:
can you set logging level to 200 in cmake? I see something like this

```
[06/09/2016 12:44:13.124] info/channel	Connection 5 | SecureChannel 5 | OpenSecureChannel: Opened SecureChannel
[06/09/2016 12:44:13.125] debug/channel	Connection 5 | SecureChannel 5 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} created
[06/09/2016 12:44:13.129] info/session	Connection 5 | SecureChannel 5 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | ActivateSession: Session activated
```
and then from another client

```
06/09/2016 12:46:26.041] info/channel	Connection 6 | SecureChannel 11 | OpenSecureChannel: Opened SecureChannel
[06/09/2016 12:46:26.041] debug/channel	Connection 6 | SecureChannel 11 | Session {B9AD21F8-21E0-CFAD-2292-A9FA5E3553B5} created
[06/09/2016 12:46:26.042] info/session	Connection 6 | SecureChannel 11 | Session {B9AD21F8-21E0-CFAD-2292-A9FA5E3553B5} | ActivateSession: Session activated

```

Root-Core, 2016-06-09:
As this happens randomly, it could take some time.. but I'm on it.

Root-Core, 2016-06-09:
Sorry, I got no time for this today, but I just got an idea about the origin of this issue. 

In this example 'abc' equals {152CA78D-6003-F3BF-BB7BEEFEEFBE}

Client 1 connects, gets the session id abc.
Client 2 connects, session id 123.
The server restarts.
Client 3 connects, gets the session id abc.
Client 4 connects, gets the session id xyz.

while (true)
{
Client 1 reconnects and wants the session abc back.
Client 3 gets kicked
Client 3 reconnects and wants the session abc back.
Client 1 gets kicked
}

This might be a totally stupid and wrong, as I am really tired right now.

Root-Core, 2016-06-10:
Here is some rather interesting logging..

```
Connection 37 | New connection over TCP from 127.0.0.1:40296
Connection 37 | SecureChannel 1 | OpenSecureChannel: Opened SecureChannel
Connection 38 | New connection over TCP from 192.168.250.72:5381
Connection 37 | SecureChannel 1 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | ActivateSession: Session activated
Connection 38 | SecureChannel 2 | OpenSecureChannel: Opened SecureChannel
Connection 40 | New connection over TCP from 192.168.250.46:58285
Connection 38 | SecureChannel 2 | Session {B7E9851D-2E4D-E71F-7107-A02AF23F5375} | ActivateSession: Session activated
Connection 40 | SecureChannel 3 | OpenSecureChannel: Opened SecureChannel
Connection 37 | SecureChannel 1 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | ActivateSession: Detach from old channel
Connection 40 | SecureChannel 3 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | ActivateSession: Session activated
```

Client 1 (127.0.0.1:40296) gets {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE}
Client 2 (192.168.250.72:5381) gets {B7E9851D-2E4D-E71F-7107-A02AF23F5375}
Client 3 (192.168.250.46:58285) gets {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE}

So, there is definitely something wrong, I guess.
As I recompiled some things, the log level got reset.. sorry.

Stasik0, 2016-06-10:
@jpfr,  is our random generator borked?

jgilje, 2016-12-20:
I ran into this issue myself. The flow described by @Root-Core [here](#issuecomment-224987392) causes two reconnecting clients to fail. I solved it by adding
```C++
#include <random>

std::random_device rd;
std::uniform_int_distribution<UA_UInt64> dist;
UA_random_seed(dist(rd));
```
before starting the OPCUA server.

Stasik0, 2016-12-20:
https://github.com/open62541/open62541/blame/243237e882f9babcdaba4ca794de84201d8c4d67/src/server/ua_server.c#L455 why is it commented out?

jpfr, 2016-12-20:
Seeding was disabled to get reproducible unit test runs.
The fuzzer I tried on the protocol did not work otherwise: http://lcamtuf.coredump.cx/afl/

I added a cmake flag to enable a deterministic RNG when needed.

commit 3b507184 (2016-12-20): seed rng if not disabled by flag, fix #632
closed 2016-12-20 by jpfr