open62541/open62541 #634: Segmentation Fault in UA_Client_Subscriptions_manuallySendPublishRequest(...);

SvenManier, 2016-06-13:
Since there is no real working subscription, I tried to make a loop to ask for changes.
```
void update(UA_Client* client) {
	int ms = time(NULL);
	int msNew;
	do {
		//sleep(5);
		UA_Client_Subscriptions_manuallySendPublishRequest(client);
		msNew = time(NULL);
		count++;
	} while (msNew - ms < 1000 * 10);
}
```

If I keep the sleep with 5 (seconds), everything is fine. But when I delete it or lower the value (i.e. 3 seconds). Im getting a Segmentation Fault
```
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7bcd240 in UA_Client_processPublishResponse (client=0x616c20, response=0x7fffffffdb40) at /home/smanier/Arbeit/open62541-0.2/src/client/ua_client_highlevel_subscriptions.c:224
224	        if(response->results[i] == UA_STATUSCODE_GOOD ||
\"Processing a publish response on subscription %u with %u notifications\",
Undefined command: "".  Try "help".
sub->SubscriptionID, response->notificationMessage.notificationDataSize);
Undefined command: "sub->SubscriptionID".  Try "help".

/* Check if the server has acknowledged any of our ACKS */
Undefined command: "".  Try "help".
// TODO: The acks should be attached to the subscription
Undefined command: "".  Try "help".
UA_Client_NotificationsAckNumber *ack, *tmpAck;
> Undefined command: "UA_Client_NotificationsAckNumber".  Try "help".
size_t i = 0;
Undefined command: "size_t".  Try "help".
LIST_FOREACH_SAFE(ack, &client->pendingNotificationsAcks, listEntry, tmpAck) {
Undefined command: "LIST_FOREACH_SAFE".  Try "help".
if(response->results[i] == UA_STATUSCODE_GOOD ||
> > > > > > > > > > > > > > > > 
```

jpfr, 2016-06-13:
We **do have** working subscriptions.
The server implementation matches the spec and behaves as expected.
It is just that the client has no own thread as of v0.2. So it cannot send publish requests on its. Hence the `manuallySendPublishRequest` function.
That will change in future versions.

I see you are using `gdb` for debugging. That's good.
Here's a tutorial with more information: http://www.unknownroad.com/rtfm/gdbtut/
Can you supply the following information about the issue?

- backtrace (`bt` command)
- call arguments (`info args`)
- local variable stat (`info locals`)

SvenManier, 2016-06-14:
Thanks for the tutorual, Im quite new using a debugger out of Eclipse, hope this is what you need

```
(gdb) backtrace
#0  0x00007ffff7bcd240 in UA_Client_processPublishResponse (client=0x616c20, response=0x7fffffffdd50) at /home/smanier/Arbeit/open62541-0.2/src/client/ua_client_highlevel_subscriptions.c:224
#1  0x00007ffff7bcd5f2 in UA_Client_Subscriptions_manuallySendPublishRequest (client=0x616c20) at /home/smanier/Arbeit/open62541-0.2/src/client/ua_client_highlevel_subscriptions.c:294
#2  0x000000000040117a in testThread (client=0x616c20) at ../src/OPC_Client_Tutorial.cpp:41
#3  0x000000000040132b in archive_main_subscribe () at ../src/OPC_Client_Tutorial.cpp:87
#4  0x00000000004011bb in main () at ../src/OPC_Client_Tutorial.cpp:49
```
```
(gdb) info args
client = 0x616c20
response = 0x7fffffffdd50
```
```
(gdb) info locals
sub = 0x617400
ack = 0x6183c0
tmpAck = 0x0
i = 0
msg = 0x0
```

Stasik0, 2016-06-14:
@SvenManier hey, is it also possible to see the full code (maybe just post it into a gist)? I saw a "thread" function, are you using a multi-threaded environment?

Stasik0, 2016-06-14:
> I was experimenting with threads but don't use them in this case
>
> static void theAnswerChanged(UA_UInt32 monitoringId, UA_DataValue *value, void *context) {
>     int* i = void_to_int(value->value.data);
>
>     if (globalValue != *i) {
>         globalValue = *i;
>         count++;
>         printf("CHANGE! - %d\n", globalValue);
>         printf("Count = %d\n", count);
>     }
> }
>
> void testThread(UA_Client* client) {
>     int ms = time(NULL);
>     int msNew;
>     do {
>         sleep(1);
>         UA_Client_Subscriptions_manuallySendPublishRequest(client);
>         msNew = time(NULL);
>         count++;
>     } while (msNew - ms < 1000 * 10);
> }
>
> int main() {
> [... 46 lines cut ...]
>     UA_Client_delete(client);
>
>     printf("Finished. End.\n"); //XXX DEBUG
>     return 0;
> }

which server are you using for your tests?

SvenManier, 2016-06-14:
I was using the FreeOpcUa-Server
Tested it with your server example and now its working...
Still a curious

Stasik0, 2016-06-14:
it is still a bug, since our software is not supposed to crash ;)

Stasik0, 2016-06-15:
@SvenManier does the fix help you?

SvenManier, 2016-06-15:
Seems like the hotfix was a success, no more SIGSEGV
Thank you =)

closed 2016-06-15 by Stasik0