open62541/open62541 #636: Events management

AlexSTI, 2016-06-14:
I'm trying to port the library to and embedded Platform based on CortexM4. The initial results are very good.
I done my first trials with the 0.1 and now the 0.2.0 is really exciting.
Now the wall were I stopped on: I'm trying to manage Events as described in the chapter 9...
I feel really lost on those.
Can we dicuss about that?
Can I offer my help on that?

Stasik0, 2016-06-15:
hey @AlexSTI, the short answer is - we got no events yet. AFAIK subscriptions are just a "kind" of events without bubbling, so probably you can start on these?

jpfr, 2016-06-15:
We need to be careful about the wording here.

*Subscriptions* with (several) *MonitoredItems*  can be defined for *DataChanges* (node attributes changing) and *Events* (emitted from objects). Currently, we only have *DataChanges*.

@AlexSTI What are you trying to achieve? For the server, you just need to enable the subscriptions feature. For clients its not as comfortable yet, since the client doesn't have an individual thread. Have a look at the client example in the /examples folder.

AlexSTI, 2016-06-15:
Thanks a lot! In effect I was taking about Events as sourced from objects. I think this is a key point. The part 9 of the specs focus on this and is very important to manage Alarms and acks. It is the key point... if something is burning we need to manage it properly...

jpfr, 2016-06-15:
As I said, we don't have events yet. Only DataChange notifications.
I'll tag this issue for reference of future releaeses.
But this will at least take until 0.3.

Michel4OpenOpcUa, 2016-06-17:
Do you know OpenOpcUa ? 
We support OPC UA Eventing feature.

AlexSTI, 2016-06-17:
Very nice! Do you think it's bad idea to port them on open62541?                                                                                                                                                                                                                                                                                                                                        Alessandro ChiesStructura Technology & InnovationSent from iPhone 8s                                                                                                                                                                                                                Da: Michel4OpenOpcUaInviato: venerdì 17 giugno 2016 13:22A: open62541/open62541Rispondi a: open62541/open62541Cc: AlexSTI; MentionOggetto: Re: [open62541/open62541] Events management (#636)Do you know OpenOpcUa ? 
We support OPC UA Eventing feature.

Stasik0, 2016-06-17:
@Michel4OpenOpcUa is CeCill-C compatible with LGPL? Where can I have a look into your codebase?

Michel4OpenOpcUa, 2016-06-17:
OpenOpcUa is available to sponsors. You can access the project at www.openopcua.org
Our approach i quite a bit different of this one. We started our project 7 years ago with the idea as creating a reference multiplatform implementation. Today our codebase is widely use all over the world 
running, Windows, Linux and VXWorks. Targeting  prcessor and µcontroler.. Feel free to contact me for more information.

FlorianPalm, 2016-06-17:
@Michel4OpenOpcUa  "The access to the OpenOpcUa codebase require a onetime  fee", calling it open source is quite confusing. Not really compareable to this project...

Michel4OpenOpcUa, 2016-06-17:
Hello,
This is not confusing. The code is Open. Free and Open are two different words with two different meaning. don't you know ?

jpfr, 2016-06-17:
You are free to sell software + source code under CeCILL-C. However, once anybody has the binaries, he can demand to also get the full sources. And anybody who has sources can redistribute them as long as they stay CeCILL-C. That is why open source with a price is rather ... unusual.

http://www.cecill.info/licences/Licence_CeCILL-C_V1-en.html
> The Licensee is authorized to distribute true copies of the Software in Source Code or Object Code form, provided that said distribution complies with all the provisions of the Agreement and is accompanied by:
- a copy of the Agreement,
- a notice relating to the limitation of both the Licensor's warranty and liability as set forth in Articles 8 and 9,

> and that, in the event that only the Object Code of the Software is redistributed, the Licensee allows effective access to the full Source Code of the Software at a minimum during the entire period of its distribution of the Software, it being understood that the additional cost of acquiring the Source Code shall not exceed the cost of transferring the data.

I happen to have a copy of an older OpenOpcUa release.
But I won't distribute it, since I understand that you intend to make it a commercial offering.
Just be careful about the fine-print @Michel4OpenOpcUa!

Most companies in the open source space have an open source version and an additional commercial (cross-licensed) version that doesn't force their customers to open source their derivative products using the library. Or they just sell services.

Michel4OpenOpcUa, 2016-06-17:
Hello,
Thank you for this information.
Couple of important thing to know about OpenOpcUa codebase 
1- Forking is fordibben
2- I can offer acess freely to university and budgetless research institute
3- What I sell and what is commercial are services around OPC UA and OpenOpcUa
        Training, Technical support, doc, etc.

Create an industrial  quality OPC UA implementation require budget and time.
Regards
Michel

Stasik0, 2016-06-17:
@Michel4OpenOpcUa 
regarding point 1: the problem is that your current license explicitly allows forking as far as I understood
http://www.cecill.info/licences/Licence_CeCILL-C_V1-en.html

> 5.2 RIGHT OF MODIFICATION
>
>The right of modification includes the right to translate, adapt, arrange, or make any or all modifications to the Software, and the right to reproduce the resulting software. It includes, in particular, the right to create a Derivative Software.
>
>The Licensee is authorized to make any or all modification to the Software provided that it includes an explicit notice that it is the author of said modification and indicates the date of the creation thereof.

furthermore, as @jpfr correctly noticed once you distribute binaries (this is what you do over your website), you need to allow access to code base as well

> 5.3.1 DISTRIBUTION OF SOFTWARE WITHOUT MODIFICATION
>
> in the event that only the Object Code of the Software is redistributed, the Licensee allows effective access to the full Source Code of the Software at a minimum during the entire period of its distribution of the Software, it being understood that the additional cost of acquiring the Source Code shall not exceed the cost of transferring the data. 

point 2: the organization will be free to copy and distribute the source code according to the license

point 3: this is correct usage of the selected open source license, so why not just opening the source for everyone and follow the license?

Michel4OpenOpcUa, 2016-06-17:
Hello,
let agree that our licence is "based on CECILL-C".
Oganisation cannot freely distribute the code or fork it. My hope is that OpenOpcUa community member works on the same project to achieve a high quality implementation. Their are so many thing to do. There a no need for various C/C++ OPC UA implementation.  We start in 2009. We support almost everything even complex feature of OPC UA. But i'm not a salary man. I make money with my sponsors and contractors. This is why i distribute it that way. I just look at your code, this is an ok code but so far from OpenOpcUa. You have to know that i'm always open to collaboration

Regards
Michel

Stasik0, 2016-06-17:
@Michel4OpenOpcUa in this case it not clear to me what is the difference OpenOpcUa and any other commercial stack where you can also can buy the source code. 
It is very nice to hear about collaboration, however I have not really seen your code (but you have seen our) so I can not imagine how we can help you. The same is true for the other people who did not pay your "membership fee". On the other side you could contribute to our project and make money with services on it.

Michel4OpenOpcUa, 2016-06-17:
Hello,
The difference with commercial toolkit is that it's commercial and always belong to the company they create it.
As i said. I sell services. I create a reference implementation ready to use. You can extend it using add-in, no need to rebuild the codebase. Add in are for communication driver and Database connector. You can also script OpenOpcUaCoreServer in LUA. Our AddressSpace is made from UANodeSet compliant XML file. There no need to nodeset compilation. This is fast, elegant, reliable and flexible. 100% ready for existing and futur OPC UA companion specification and for Industrie 4.0. I'm sure you can imagine.
But for sure collaboration is difficult.. My codebase is huge and very different of you codebase in term of implementation. for example : Do you have user define datatype and their datatype dictionnary ?

Regards
Michel

AlexSTI, 2016-06-17:
I am realy sorry to have been started a so long unusefull discussion... Sorry with the great guys that maintain this marvellous Lib.
Let me spend only one word: dear Michel, maybe you could be one of the best codewriters in the world but when you approach opensource and you publish it as open, you have to be free to really be open. I gave a look to your fees and sorry, there is nothing about that on what you do. Yours is for shure a commercial product with a downloadable sample. Of course it will be reliable, fast, elegant, pretty and all the best but not opensource. Open source is efferts done for the community and shared with the community of hard workers. Ask linux kernels or UBoot writers if they ask for those fees to have their code... I suppose you use Linux, did you pay 15k€ to obtain that magnific SW? Proud to use open62541!
And please, don't use open source sites to insert spots for your commercial activity.
Going bak to serious things, how can I help to implements Events subscriptions? I think this is really very important to know if an alarm exists and to ack/confirm it... if something brackes down, it is useless to modify its setpoint...
Best regards to all the opensourcers

Stasik0, 2016-06-17:
@AlexSTI do you have access to the spec? You can get it from opc foundation. From what i know about events, the delivery system should be pretty similar to subscriptions i.e. their data change events (which is one special event type). So it is a good way to start is to look into the code of subscription service with its internal publishing and sampling jobs. Probably the publishing job can be simply reused with some mechanism other than the sampling job that fills the response queue.

jpfr, 2016-06-17:
All of us will profit if OPC UA succeeds as a standard. So :+1: for collaboration and making this great technology easily accessible.

@AlexSTI At some point people have to live from their work. So let's not be negative about those who do. Not everybody has such lofty research positions that allow them to write open source code on the side.

@Michel4OpenOpcUa We'll make sure to point people to your project if they require features we do not provide, professional support, and so on. Is there a list of the UA profiles/feature facets you currently support?
However, we cannot do an official endorsement from the open62541 project. Our organizations have relations to pretty much every vendor out there. I hope you understand that.

AlexSTI, 2016-06-17:
I agree jpfr! I own a company with some peoples inside... I know that at the supermarket, you can't pay with opensource code... this is because my company produces products and sells them! And we try to do it the best we can. But if you say that something is open... it MUST be 100% open, otherways it is not... we not give our schematics and our OS source because it is not open... but we say "it is closed!" and we gain money from there. Opensource is something else... 
Best regards and, again, I will give a look to the Subscriptions in detail, and I will be back soon with something to share (both in terms of suggestions than in askings...).

Michel4OpenOpcUa, 2016-06-17:
Hello,
I'm not here to sell my codebase. My codebase is not for sell. I sell services. I provide Something to help people to understand and to adopt UA.

My project is OpenSource but i need to build an economic model on the top of it. Open62541 receive sponsor don't you know that ?
I do the same. @AlexSTI  try to be open minded.... Maybe you remember the time when OpenOpcUa was totaly free. There is companies that use OpenOpcUa for real product... They paid 500€ and they create wonderfull product without any royalties... This is Open !!! 
But again Free and Open are just different. The codebase is Open not for free ;) Somebody says :
"Something free have no value" .. OpenOpcUa is Open. 
I don't really care about OSes.. I care about technology and first of all about OPC UA.
@jpfr : I support UA Standard server profile + DataAccessServer Facet + some not testable facet (CTT not ready)
@AlexSTI : I you want i can setup a gotomeeting to explain how Eventing is suppose to works with UA.
Kind regards
Michel

ichrispa, 2016-06-23:
I would like to strongly suggest to keep issues posted here focused on what this plattform is meant to do: improve and discuss open62541. I see any discussions about commercially offered alternatives, their license schemes and/or advertisement completely out of place in this issue.

@AlexSTI: As @Stasik0 has pointed out, we do currently not support event. It has been on my todo list for a long time, but unfortunately I definetely do not have the assets to implement much of anything open62541 related at this time. We could set it as a target feature for 0.3 and hopefully my workload will drop enough to come around to contributing again. 

@Michel4OpenOpcUa: Thank you for pointing out that your Stack offers this feature - if you can contribute your expiriences or parts of your codebase, feel free to support our open source project. If you have nothing to contribute to the issue at hand, please refrain from posting. This project has been around long enough to understand how open source projects can be commercialized and viable, diminuitive posts of the form "don't you know how it works" are neither necessary nor relevant to this discussion plattform.

symgyyb, 2016-08-26:
hello,
        When I think the new version will be released ? What features will increase ? Where can we see the latest news ?
       Hope for your reply,thanks.

ichrispa, 2016-08-26:
Definetely once I get around to coding again. But as with all open source projects, there is no timetable.

jpfr, 2017-06-19:
Duplicate of #245

closed 2017-06-19 by jpfr