open62541/open62541 #644: Sending large blocks ( > 4k) fails on ARM

Andreas-N, 2016-06-21:
Hi,
I'm trying to implement a OPC UA server on a small device running linux on an arm processor. Reading an UA_ByteString object with UAExpert fails, when the size of the block gets larger than 4k Byte (same happens with arrays larger than 4K). The bug seams to be in the socket_write method in plugins/networklayer_tcp.c in line 85. When the send call does not transmit the complete block (n < buf->length) than in the next loop pass through the bytes already transmitted will be send again. In addition the send call can fail with EMSGSIZE, when the block cannot be transmitted atomically through the network layer (See man send).
I suggest the following modification to resolve these problem:
```c
        do {
          // Send smaller chunks to avoid EMSGSIZE
          size_t bytes_to_send = buf->length - nWritten >  1024 ? 1024 : buf->length - nWritten;
#ifdef _WIN32
            // n = send((SOCKET)connection->sockfd, (const char*)buf->data, buf->length, 0);
            n = send((SOCKET)connection->sockfd, (const char*)buf->data + nWritten, bytes_to_send, 0);
            const int last_error = WSAGetLastError();
            if(n < 0 && last_error != WSAEINTR && last_error != WSAEWOULDBLOCK) {
                connection->close(connection);
                socket_close(connection);
                UA_ByteString_deleteMembers(buf);
                return UA_STATUSCODE_BADCONNECTIONCLOSED;
            }
#else
            // n = send(connection->sockfd, (const char*)buf->data, buf->length, MSG_NOSIGNAL);
            n = send(connection->sockfd, (const char*)buf->data + nWritten, bytes_to_send, MSG_NOSIGNAL);
            if(n == -1L && errno != EINTR && errno != EAGAIN) {
                connection->close(connection);
                socket_close(connection);
                UA_ByteString_deleteMembers(buf);
                return UA_STATUSCODE_BADCONNECTIONCLOSED;
            }
#endif
        } while(n == -1L);
``` 
Regards, Andreas

jpfr, 2016-06-21:
Great find.
Fixed.

I guess in the past everbody used either
- an OS that split the TCP stream in smaller packets automatically, or
- set the maximum size of OPC UA chunks to a small enough value (OPC UA messages can be split into chunks and demultiplexed on the receiving side)

closed 2016-06-21 by jpfr

Stasik0, 2016-06-21:
can we add some automatic 'probing' for EMSGSIZE?

commit 87393df3 (2016-06-21): fix #644; sending tcp messages in blocks