open62541/open62541 #645: using malloc, free etc.

NetoLe, 2016-06-21:
Hello! I liked your idea of mapping the Allocation-functions to system-specific functions.

But it seems that newest commits don't use this functionality. I saw at least 2 mallocs which are no UA_malloc in the latest commits. That broke my server, sure it is fast to fix, but I would recommend to stay with the original idea. As well I would say you should use them as well in the tcp_networklayer etc. because I'm sure that many people use this component as well (so for me).

best regards

FlorianPalm, 2016-06-21:
The system depended funcitons were accidently introduced. I think @Stasik0 is fixing it atm.

Stasik0, 2016-06-21:
@NetoLe is it this commit that you mean? https://github.com/open62541/open62541/commit/984342b057440a91fe7686ee1eea4c9d9cc4eb8a or are there some more mallocs that are used outside of the tests?

NetoLe, 2016-06-21:
yes those are the ones I needed to fix. But I didn't watch the whole code to be honest.

Stasik0, 2016-06-21:
i will change it back asap, happened by an accident as @FlorianPalm correctly noticed

jpfr, 2016-06-21:
The changes in ua_types.h should be reverted.
The difference between `UA_STRING` and `UA_STRING_ALLOC` is precisely that `UA_STRING` does not allocate memory. And `UA_STRING_ALLOC` uses the internal malloc as @NetoLe requested.

@ichrispa Please use `UA_STRING_ALLOC` if you C++-compiler doesn't like the pointer magic.

commit bc1291e9 (2016-06-21): fix #645
closed 2016-06-21 by jpfr