open62541/open62541 #646: Getting every change

SvenManier, 2016-06-23:
Hello again,
im still working with your library and I have a short question.
Situation:
Server - the.answer = 1
Client - sendPhblishRequest, the.answer = 1
Server - the.answer = 2
Server - the.answer = 3
Server - the.answer = 4
Client - sendPhblishRequest, the.answer = 4
Server - the.answer = 5
Is there a way to get (really) ALL changes the server does?

Stasik0, 2016-06-23:
the currently used subscription sampling mechanism is limited to a sampling rate, currently we have no event flow mechanisms enabled (neither in the server nor on the opc ua service side), so this is currently not possible.

p.s.: well if you really need this feature, I can imagine that you can change the source code and tie the change event or its write to the subscription somehow.

SvenManier, 2016-06-23:
Allright, thank you

closed 2016-06-23 by Stasik0

jpfr, 2016-06-23:
@SvenManier You can change the sampling interval of the MonitoredItem independent from the publication interval of the subscription.

Just take a sampling interval that is low enough (say 50ms). You will then get multiple notifications with every publish response. The load on the CPU is usually negligible.

tijo40, 2016-11-04:
Hello @jpfr ,

I have a similar question. 

I have added a job at the Server which calls every 5000 milliSeconds. When the job runs, it will update variable node at the Server multiple times. The time interval between each write operation is set as 100 milliSeconds(for test purpose). At the Client side, subscription is created to monitor the variable Node.  But with every publish request, i am getting the last written value from the Job.  I have set "minimumSamplingInterval" as 50 when the variable node was created. Tried reducing  "minimumSamplingInterval" to 10, still not able to retrieve the intermediate values. 

I am using the default "UA_SubscriptionSettings_standard" from "v0.2.0-RC1" in Windows 7 platform.

Is there any Option to get the history of Attribute values?

jpfr, 2016-11-04:
Hello @tijo40 ,

depending on what the client requests, the actual sampling interval might be larger than the minimumSamplingInterval. That's only the lower bound.

Also, the current 0.2 branch has some improvements for monitoreditems over the 0.2-rc1 release.

tijo40, 2016-11-07:
Hello @jpfr ,

I was using single file distribution of "open62541" for my build from v0.2.0-RC1. 
Build IDE : QT, Compiler : mingw32, language : c++, 

I tried generating single header & source file for Version 0.2 by setting UA_ENABLE_AMALGAMATION ON. While configuring cmake, selected "MinGW Makefiles". But when i check in the binary path of cmake, I am not able to find single file distribution.  

Could you please let me know where i am wrong with generating single files for Version 0.2?