open62541/open62541 #655: stress test causes loop (in client or server)

StalderT, 2016-06-24:
Hello,

I add this to server.c :
```
static UA_StatusCode
readTest(void *handle, const UA_NodeId nodeId, UA_Boolean sourceTimeStamp,
         const UA_NumericRange *range, UA_DataValue *value)
{
    if(range) {
        value->hasStatus = UA_TRUE;
        value->status = UA_STATUSCODE_BADINDEXRANGEINVALID;
        return UA_STATUSCODE_GOOD;
    }   

    UA_UInt32 val = (UA_UInt32)10;
    UA_DateTime currentTime = UA_DateTime_now();
    UA_Variant_setScalarCopy(&value->value, &val, &UA_TYPES[UA_TYPES_UINT32]);
    value->hasValue = UA_TRUE;

    if(sourceTimeStamp) {
        value->hasSourceTimestamp = UA_TRUE;
        value->sourceTimestamp = currentTime;
    }   

    return UA_STATUSCODE_GOOD;
}
```
and
```
    for(UA_UInt32 i = 0; i < 5000; i++) {
        UA_DataSource dateDataSource_test = (UA_DataSource) {
            .handle = (void*)i, .read = readTest, .write = NULL
        };
        UA_VariableAttributes v_attr_test;
        UA_VariableAttributes_init(&v_attr_test);
        char name[100];
        snprintf(name, 100, "TEST%04d", i);
        v_attr_test.description = UA_LOCALIZEDTEXT("en_US", name);
        v_attr_test.displayName = UA_LOCALIZEDTEXT("en_US", name);
        v_attr_test.accessLevel = UA_ACCESSLEVELMASK_READ;
        const UA_QualifiedName dateName_test = UA_QUALIFIEDNAME(1, name);
        UA_NodeId dataSourceId_test;
        UA_Server_addDataSourceVariableNode(server, UA_NODEID_NUMERIC(1, 1000+i),
                                            UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                                            UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES), dateName_test,
                                            UA_NODEID_NULL, v_attr_test, dateDataSource_test, &dataSourceId_test);
    }
```
I add this to client.c :
```
    for(int i=0;i<5000; i++)
    {
    UA_NodeId monitorThis = UA_NODEID_NUMERIC(1, 1000+i);
    UA_UInt32 monId = 0;
    UA_Client_Subscriptions_addMonitoredItem(client, subId, monitorThis, UA_ATTRIBUTEID_VALUE,
                                             &handler_TheAnswerChanged, (void*)i, &monId);
    if (monId)
        printf("Monitoring i=%d, id %u\n", i, subId);
    /* The first publish request should return the initial value of the variable */
    }
```
and remove : 
```
    /* Browse some objects */
    printf("Browsing nodes in objects folder:\n");
    UA_BrowseRequest bReq;
    UA_BrowseRequest_init(&bReq);
    bReq.requestedMaxReferencesPerNode = 0;
    bReq.nodesToBrowse = UA_BrowseDescription_new();
    bReq.nodesToBrowseSize = 1;
    bReq.nodesToBrowse[0].nodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER); /* browse objects folder */
    bReq.nodesToBrowse[0].resultMask = UA_BROWSERESULTMASK_ALL; /* return everything */
    UA_BrowseResponse bResp = UA_Client_Service_browse(client, bReq);
    printf("%-9s %-16s %-16s %-16s\n", "NAMESPACE", "NODEID", "BROWSE NAME", "DISPLAY NAME");
    for (size_t i = 0; i < bResp.resultsSize; ++i) {
        for (size_t j = 0; j < bResp.results[i].referencesSize; ++j) {
            UA_ReferenceDescription *ref = &(bResp.results[i].references[j]);
            if(ref->nodeId.nodeId.identifierType == UA_NODEIDTYPE_NUMERIC) {
                printf("%-9d %-16d %-16.*s %-16.*s\n", ref->browseName.namespaceIndex,
                       ref->nodeId.nodeId.identifier.numeric, (int)ref->browseName.name.length,
                       ref->browseName.name.data, (int)ref->displayName.text.length,
                       ref->displayName.text.data);
            } else if(ref->nodeId.nodeId.identifierType == UA_NODEIDTYPE_STRING) {
                printf("%-9d %-16.*s %-16.*s %-16.*s\n", ref->browseName.namespaceIndex,
                       (int)ref->nodeId.nodeId.identifier.string.length,
                       ref->nodeId.nodeId.identifier.string.data,
                       (int)ref->browseName.name.length, ref->browseName.name.data,
                       (int)ref->displayName.text.length, ref->displayName.text.data);
            }
            /* TODO: distinguish further types */
        }
    }
    UA_BrowseRequest_deleteMembers(&bReq);
    UA_BrowseResponse_deleteMembers(&bResp);

    /* Same thing, this time using the node iterator... */
    UA_NodeId *parent = UA_NodeId_new();
    *parent = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_Client_forEachChildNodeCall(client, UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                                   nodeIter, (void *) parent);
    UA_NodeId_delete(parent);

```
I start the server and then I start the client, all works fine.

If I start the server, then start UA_Expert and connect to the server, then start the client, the client loop in UA_Client_Subscriptions_manuallySendPublishRequest(client);

I don't know if the problem come from the client or the server.

jpfr, 2016-06-24:
Normally, this can be easily verfied.
Run client and server in a debugger. Break when you are in the "loop" and show us the call stack.

StalderT, 2016-06-24:
Hello,

For information the code for server are : 

```
static UA_StatusCode
readTest(void *handle, const UA_NodeId nodeId, UA_Boolean sourceTimeStamp,
         const UA_NumericRange *range, UA_DataValue *value)
{
    if(range) {
        value->hasStatus = UA_TRUE;
        value->status = UA_STATUSCODE_BADINDEXRANGEINVALID;
        return UA_STATUSCODE_GOOD;
    }   

    UA_UInt32 val = (UA_UInt32)10+rand();
    UA_DateTime currentTime = UA_DateTime_now();
    UA_Variant_setScalarCopy(&value->value, &val, &UA_TYPES[UA_TYPES_UINT32]);
    value->hasValue = UA_TRUE;

    if(sourceTimeStamp) {
        value->hasSourceTimestamp = UA_TRUE;
        value->sourceTimestamp = currentTime;
    }   

    return UA_STATUSCODE_GOOD;
}
```
and not : 

```
static UA_StatusCode
readTest(void *handle, const UA_NodeId nodeId, UA_Boolean sourceTimeStamp,
         const UA_NumericRange *range, UA_DataValue *value)
{
    if(range) {
        value->hasStatus = UA_TRUE;
        value->status = UA_STATUSCODE_BADINDEXRANGEINVALID;
        return UA_STATUSCODE_GOOD;
    }   

    UA_UInt32 val = (UA_UInt32)10;
    UA_DateTime currentTime = UA_DateTime_now();
    UA_Variant_setScalarCopy(&value->value, &val, &UA_TYPES[UA_TYPES_UINT32]);
    value->hasValue = UA_TRUE;

    if(sourceTimeStamp) {
        value->hasSourceTimestamp = UA_TRUE;
        value->sourceTimestamp = currentTime;
    }   

    return UA_STATUSCODE_GOOD;
}
```
ps output : 

```
root      1677  0.0  0.5  15308 11540 pts/0    -    14:43   0:00 gdb server
root         -  0.0    -      -     - -        S    14:43   0:00 -
root      1679 66.1  0.3   7920  7160 pts/0    -    14:43   1:29 /data/server
root         - 66.1    -      -     - -        R+   14:43   1:29 -
root      1683  0.0  0.5  15344 11812 pts/1    -    14:43   0:00 gdb client
root         -  0.0    -      -     - -        S    14:43   0:00 -
root      1690 10.4  0.0   2572  1916 pts/1    -    14:43   0:09 /data/client
root         - 10.4    -      -     - -        S+   14:43   0:09 -

```

How can I break and show call stack ?

StalderT, 2016-06-24:
I found.

Server side : 

```
^C
Program received signal SIGINT, Interrupt.
0xb7fbaf0f in UA_Subscription_publishCallback (server=0x804f038, sub=0x83897b0) at /data/linux/open62541/src/build/open62541.c:21381
21381	            TAILQ_FOREACH_SAFE(qv, &mon->queue, listEntry, qv_tmp) {
(gdb) backtrace
#0  0xb7fbaf0f in UA_Subscription_publishCallback (server=0x804f038, sub=0x83897b0) at /data/linux/open62541/src/build/open62541.c:21381
#1  0xb7fbc140 in Service_Publish (server=0x804f038, session=0x8499f50, request=0xbfffece0, requestId=52341) at /data/linux/open62541/src/build/open62541.c:21843
#2  0xb7fac95e in processRequest (channel=0x8499de8, server=0x804f038, requestId=52341, msg=0xbfffeedc) at /data/linux/open62541/src/build/open62541.c:14889
#3  0xb7facd21 in processMSG (connection=0x8390708, server=0x804f038, messageHeader=0xbfffefcc, msg=0x8469e68, offset=0xbfffefd4) at /data/linux/open62541/src/build/open62541.c:14978
#4  0xb7facf84 in UA_Server_processBinaryMessage (server=0x804f038, connection=0x8390708, msg=0x8469e68) at /data/linux/open62541/src/build/open62541.c:15058
#5  0xb7fad68b in processJobs (server=0x804f038, jobs=0x8469e60, jobsSize=1) at /data/linux/open62541/src/build/open62541.c:15308
#6  0xb7fae095 in UA_Server_run_iterate (server=0x804f038, waitInternal=true) at /data/linux/open62541/src/build/open62541.c:15923
#7  0xb7fae1e0 in UA_Server_run (server=0x804f038, running=0x804d76c <running>) at /data/linux/open62541/src/build/open62541.c:15970
#8  0x0804bd2a in main ()
```

or 

```
Program received signal SIGINT, Interrupt.
0xb7fdbad0 in __kernel_vsyscall ()
(gdb) backtrace
#0  0xb7fdbad0 in __kernel_vsyscall ()
#1  0xb7ef1eee in send () from /lib/libc.so.6
#2  0xb7fbd03a in socket_write (connection=0x829b7e8, buf=0xbfffea50) at /data/linux/open62541/src/build/open62541.c:22328
#3  0xb7fa3949 in UA_SecureChannel_sendChunk (ci=0xbfffea18, dst=0xbfffea50, offset=290) at /data/linux/open62541/src/build/open62541.c:12736
#4  0xb7fa3b34 in UA_SecureChannel_sendBinaryMessage (channel=0x829bcb8, requestId=12640, content=0x83b1468, contentType=0x804e298 <UA_TYPES+2808>) at /data/linux/open62541/src/build/open62541.c:12801
#5  0xb7fbb201 in UA_Subscription_publishCallback (server=0x804f038, sub=0x829bf70) at /data/linux/open62541/src/build/open62541.c:21426
#6  0xb7fbc140 in Service_Publish (server=0x804f038, session=0x829c578, request=0xbfffece0, requestId=12640) at /data/linux/open62541/src/build/open62541.c:21843
#7  0xb7fac95e in processRequest (channel=0x829bcb8, server=0x804f038, requestId=12640, msg=0xbfffeedc) at /data/linux/open62541/src/build/open62541.c:14889
#8  0xb7facd21 in processMSG (connection=0x829b7e8, server=0x804f038, messageHeader=0xbfffefcc, msg=0x82b1318, offset=0xbfffefd4) at /data/linux/open62541/src/build/open62541.c:14978
#9  0xb7facf84 in UA_Server_processBinaryMessage (server=0x804f038, connection=0x829b7e8, msg=0x82b1318) at /data/linux/open62541/src/build/open62541.c:15058
#10 0xb7fad68b in processJobs (server=0x804f038, jobs=0x82b1310, jobsSize=1) at /data/linux/open62541/src/build/open62541.c:15308
#11 0xb7fae095 in UA_Server_run_iterate (server=0x804f038, waitInternal=true) at /data/linux/open62541/src/build/open62541.c:15923
#12 0xb7fae1e0 in UA_Server_run (server=0x804f038, running=0x804d76c <running>) at /data/linux/open62541/src/build/open62541.c:15970
#13 0x0804bd2a in main ()
```

Client side
```
^C
Program received signal SIGINT, Interrupt.
0xb7fdbad0 in __kernel_vsyscall ()
(gdb) backtrace
#0  0xb7fdbad0 in __kernel_vsyscall ()
#1  0xb7ef1d6e in recv () from /lib/libc.so.6
#2  0xb7fbd1be in socket_recv (connection=0x804c00c, response=0xbffff5b8, timeout=5000) at /data/linux/open62541/src/build/open62541.c:22398
#3  0xb7fb7adc in __UA_Client_Service (client=0x804c008, r=0xbffff6d8, requestType=0x804b0f8 <UA_TYPES+3960>, response=0xbffff61c, responseType=0x804ac78 <UA_TYPES+2808>) at /data/linux/open62541/src/build/open62541.c:19870
#4  0xb7f9cfd4 in UA_Client_Service_publish (client=0x804c008, request=...) at /data/linux/open62541/src/build/open62541.h:6275
#5  0xb7fbcf47 in UA_Client_Subscriptions_manuallySendPublishRequest (client=0x804c008) at /data/linux/open62541/src/build/open62541.c:22226
#6  0x08049225 in main ()

```

The server seem to continually send values.

jpfr, 2016-06-26:
From what you posted, the server is responding to a publish request. As it should do.
It seems you are calling `UA_Client_Subscriptions_manuallySendPublishRequest` in a loop.

jpfr, 2016-06-26:
Maybe reduce the UA_LOGLEVEL to 100 to see if you are really blocking or just calling the same service again and again.

StalderT, 2016-06-26:
No, client.c only call one time UA_Client_Subscriptions_manuallySendPublishRequest (maybe two times) 

I stop the client, but server continue to send values (client and ua_expert are stopped) : 

```
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Subscription 1 | MonitoredItem 3778 | Sampling the value
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Subscription 1 | MonitoredItem 3779 | Sampling the value
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Subscription 1 | MonitoredItem 3780 | Sampling the value
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Subscription 1 | MonitoredItem 3781 | Sampling the value
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Subscription 1 | MonitoredItem 3782 | Sampling the value
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Subscription 1 | MonitoredItem 3783 | Sampling the value
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Subscription 1 | MonitoredItem 3784 | Sampling the value
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Subscription 1 | MonitoredItem 3785 | Sampling the value
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Subscription 1 | MonitoredItem 3786 | Sampling the value
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Subscription 1 | MonitoredItem 3787 | Sampling the value
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Subscription 1 | MonitoredItem 3788 | Sampling the value
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Subscription 1 | MonitoredItem 3789 | Sampling the value
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Subscription 1 | MonitoredItem 3790 | Sampling the value
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/26/2016 14:27:34.184] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Subscription 1 | MonitoredItem 3791 | Sampling the value

```

With UA_LOGLEVEL to 100 the problem always occur.

StalderT, 2016-06-27:
For info, if I stop the client after a while the server stop the loop and say : 
```
[06/27/2016 12:28:09.564] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Subscription 1 | MonitoredItem 3921 | Do not sample an unchanged value
[06/27/2016 12:28:09.564] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/27/2016 12:28:09.564] debug/session	Connection 0 | SecureChannel 0 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Subscription 1 | MonitoredItem 3922 | Do not sample an unchanged value
[06/27/2016 12:28:09.564] debug/session	Session with token 355248013 has timed out and is removed
```

jpfr, 2016-06-27:
You create a monitoreditem with a given sampling interval.
The server is supposed to regularly check the value for changes. Not only when you send a publish request.

That's just how monitoreditems work.
Can we close?

StalderT, 2016-06-27:
Thank you for explaining. But I don't understand why UA_Client_Subscriptions_manuallySendPublishRequest never exit

jpfr, 2016-06-27:
Ah okay. Now I understand your issue.
I'll check that with the code example you provided.

StalderT, 2016-06-28:
I have made some more tests.

I call 
```
   UA_SubscriptionSettings UA_SubscriptionSettings_custom = { 
        .requestedPublishingInterval = 5000.0,
        //.requestedPublishingInterval = 500.0,
        .requestedLifetimeCount = 10000,
        .requestedMaxKeepAliveCount = 1,
        .maxNotificationsPerPublish = 10, 
        .publishingEnabled = true,
        .priority = 0 
    };  

    UA_Client_Subscriptions_new(client, UA_SubscriptionSettings_custom, &subId);
```
instead of 
```
UA_Client_Subscriptions_new(client, UA_SubscriptionSettings_standard, &subId);
```

and all works fine.

Maybe if the time to create / process the message is longer than requestedPublishingInterval then the server sends several times the change in the same query and never end ????

nikita158, 2016-10-19:
Hi guys!

Quick/dirty fix for problem is changing:
```C
static UA_DateTime
processRepeatedJobs(UA_Server *server, UA_DateTime current) {
...
        /* Set the time for the next execution */
        rj->nextTime += (UA_Int64)rj->interval;
        if(rj->nextTime < current)
            rj->nextTime = current;
...
}
```
to
```C
static UA_DateTime
processRepeatedJobs(UA_Server *server, UA_DateTime current) {
...
        /* Set the time for the next execution */
        rj->nextTime += (UA_Int64)rj->interval;
        if(rj->nextTime < current)
            rj->nextTime = UA_DateTime_nowMonotonic();
...
}
```
@jpfr I could do a nicer pull in a few days. Should I?

nikita158, 2016-10-19:
for me seems that when `(rj->nextTime < current)` we should warn to log...

jpfr, 2016-10-28:
@nikita158 
@StalderT 

the issue is found and fixed in 0b2cec1. See the exaplanation in the commit message.
Ok to close?

StalderT, 2016-10-28:
For me you can close. 

Thank you.

closed 2016-10-28 by jpfr