open62541/open62541 #657: enable generation of monitoreditems with somes "BADxxx"

StalderT, 2016-06-27:
That would be great to enable generation of monitoreditems with some "BADxxx"

see https://github.com/open62541/open62541/commit/db0c995daa2e2e0847b127fff3ddf597110a9e06 comments

I suggest : 

UA_STATUSCODE_BADRESOURCEUNAVAILABLE 
UA_STATUSCODE_BADCOMMUNICATIONERROR

closed 2016-06-28 by jpfr

Stasik0, 2016-06-28:
is there something in the standard?
@jpfr cannot we use some "own" statuscodes that do not colliede with ua ones?

jpfr, 2016-06-28:
@Stasik0 We could indeed. But it will be really hard to ensure that nonstandard codes don't leak out to clients.

Here's the table from part 4. There are some reserved bits currently unused.
[image]

Stasik0, 2016-06-28:
Or just to rework the function to use a distinct return value.

commit 4ab6abe1 (2016-06-28): fix #657; allow some statuscodes when creating monitoreditems