open62541/open62541 #658: Client_Service with NULL-Pointer

TorbenD, 2016-06-27:
Hi,
there is no check if the handed over pointer is a NULL-Pointer or not. The function just hand over the pointer to other functions which are crashing because of the NULL-Pointer.
Chears
Torben

jpfr, 2016-06-27:
You refer to the "client" argument?

TorbenD, 2016-06-27:
yes

jpfr, 2016-06-28:
I'm inclined to say *wontfix*.

We often use "object emulation", where functions are defined explicitly for a certain datatype.

`UA_sometype_somefunc(UA_sometype *self, other args)`

To me, this cannot be expected to work if self is NULL.
If we add checks for the "self" argument, we'd have to do it consistently everywhere.
I'd rather leave a notice in the docs that all pointer are assumed non-null if not otherwise documented.

Anybody else of the core group with an opinion?

Stasik0, 2016-06-28:
+1 for _wontfix_ for the sake of efficiency ;)

closed 2016-06-29 by Stasik0