open62541/open62541 #660: Retransmission queue keeps growing when no acknowledgement is received

Andreas-N, 2016-06-30:
Hi,
I have implemented an OPC UA Server with your stack on an arm device. Some nodes contain a bytestring with a size between 10k and 60k. When subscribing to these notes  with UAExpert, UAExpert occasionally  sends a republish request ("Calling Republish with sequence number xxx"). Sometimes these requests fail and I see an error: "Call Republish failed. (ret=BadEndOfStream) in the log of UAExpert. When this happens, the notification message never gets acknowledged and remains in the retransmission queue. 
Is their a way, which conforms to the OPC specification to expires those messages in retransmission queue. 
Unfortunately I am not so familiar with the OPC UA protocol to solve it myself.

Regards Andreas

jpfr, 2016-06-30:
Hello @Andreas-N 

1. We should try to not see BadEndOfStream in UAExpert. That's perhaps due to the large bytestrings. We'll check.
2. We have `queueSizeLimits` in the server config for monitored items. But that is the queue for values that were sampled and not yet sent out. We notify the client which transmission he did not acknowledge. So there is no reason why UAExpert does not retry the resubmission.
Anyhow, we should protect against a malicious client trying to grow the retransmission queue indefinitely. But discarding messages from the retransmission queue is a very hot topic. Many users expect to get a complete history of samples even when the connectivity fails for some amount of time.

Andreas-N, 2016-06-30:
I agree, this is a pretty difficult topic to solve. Maybe a configurable time out and/or a queue limit (should be individual for each subscription because of the different message sizes or even related to memory usage of the queues instead of limiting the number of messages queued) would solve the problem for most users. Especially on embedded devices, like ours, where memory is limited, we cannot hold to many messages in the retransmission queue.

jpfr, 2016-12-02:
Implemented in 243237e882f9babcdaba4ca794de84201d8c4d67

closed 2016-12-02 by jpfr