open62541/open62541 #661: endpointDescriptions in the case of multiple networklayers

tiisch, 2016-06-30:
Hey,

in ua_server_worker.c within the implementation of the UA_Server_run_startup function the code snippet shown below leads to problems I guess:

- the outer for loop iterates over all networklayer instances
- the inner for loop (commented out) iterates over all endpointDescriptions of the server object and copies the discoveryUrl of the currently selected networklayer instance into the endpointUrl of the endpointDescription
- in the end, the discoveryUrl of the networklayer instance which is choosen in the last run of the outer loop will appear as the endpointUrl of ALL endpontDescriptions (because any subsequent call of the copy-method overwrites any preceding call)

[image]

Is this the intended behaviour? I propose to change the code as shown in the snippet above (assuming every NL instance has one associated endpoint, which is the unsecured binary one right now) or am I wrong?

Best regards

FlorianPalm, 2016-06-30:
There are several issues at the moment with the endpointDescriptions. Within the applications branch we are testing some use caes. I think we are going to fix those issues within the next few weeks

jpfr, 2016-06-30:
I would kick out the commented code and move the functionality to `Service_GetEndpoints` in ua_services_discovery.h. There, it is easy to loop over the networklayers and clone the relevant endpoints with an adjusted endpointurl.

commit e15cdfac (2016-07-01): fix #661, take the endpointurls from the networklayer when required
closed 2016-07-01 by jpfr