open62541/open62541 #666: Multiple subscriptions broken?

SvenManier, 2016-07-06:
When I try to run more than one subscription at once, it seems like just the last sub will do it.
With this setup I only get changes for ServerMaschine #2
```
UA_Client* client = UA_Client_new(UA_ClientConfig_standard);
UA_Client_connect(client, SERVER_ADDRESS.c_str());

// ##########

/* Create a subscription */
UA_UInt32 subId = 0;
UA_Client_Subscriptions_new(client, UA_SubscriptionSettings_standard, &subId);
if (subId)
	printf("Create subscription succeeded, id %u\n", subId);
/* Add a MonitoredItem */
UA_NodeId monitorThis = UA_NODEID_STRING(0, (char*) "ServerMaschine #1");
UA_UInt32 monId = 0;
UA_Client_Subscriptions_addMonitoredItem(client, subId, monitorThis, UA_ATTRIBUTEID_VALUE, &theAnswerChanged, NULL, &monId);
if (monId)
	printf("Monitoring 'the.answer', id %u\n", subId);
/* The first publish request should return the initial value of the variable */
UA_Client_Subscriptions_manuallySendPublishRequest(client);

// ##########

/* Create a subscription */
UA_UInt32 subId1 = 1;
UA_Client_Subscriptions_new(client, UA_SubscriptionSettings_standard, &subId1);
if (subId1)
	printf("Create subscription succeeded, id %u\n", subId1);
/* Add a MonitoredItem */
UA_NodeId monitorThis1 = UA_NODEID_STRING(0, (char*) "ServerMaschine #2");
UA_UInt32 monId1 = 1;
UA_Client_Subscriptions_addMonitoredItem(client, subId1, monitorThis1, 
A_ATTRIBUTEID_VALUE, &theAnswerChanged, NULL, &monId1);
if (monId1)
	printf("Monitoring 'the.answer', id %u\n", subId1);
/* The first publish request should return the initial value of the variable */
UA_Client_Subscriptions_manuallySendPublishRequest(client);

// ##########

char answ;
do {
	cout << "Read? (y/n): ";
	cin >> answ;
	try {
		UA_Client_Subscriptions_manuallySendPublishRequest(client);
	} catch (OpcClientWrapperException &e) {
		cout << e.what();
	}
} while (answ != 'n');
```
I also tried subId1 and monId1 == 0 at beginning and first adding both subs, then publish them. None of them worked

jpfr, 2016-07-07:
Hello @SvenManier,

1. The subscription id and the monitoring id are assigned by the server. You have no influence over them.
2. You can add both monitoreditems to just one subscription, that simplifies the code.
3. You use subId at some points when you mean monId.

With these changes, your example works on my box as intended.

SvenManier, 2016-07-07:
Can you tell me where exactly I use wrong subIDs? Im using the code from your example client (https://github.com/open62541/open62541/blob/0.2/examples/client.c , line: 106)

jpfr, 2016-07-07:
```c
if (monId)
    printf("Monitoring 'the.answer', id %u\n", subId);
if (monId1)
    printf("Monitoring 'the.answer', id %u\n", subId1);
```

That's not wrong per se. But it might lead to wrong conclusions during debugging.
Also, since you are using different subscriptions, you might get the same monitoringid for the two monitoreditems.
So maybe everything works alright but your output makes it look as if only one subscription is active.

SvenManier, 2016-07-11:
Mmh, no
I did run the debugger, both times I get the monitoringId = 0 and I only get the latest sub (I change the nodes with UA-Expert)
Could there be something wrong with my server? With this method I create the nodes

```
UA_StatusCode createNode(UA_Server* server, string nodeName, string defaultNodeValue, int nodeIndex) {
	/* 1) Define the attribute of the myInteger variable node */
	UA_VariableAttributes attr;
	UA_VariableAttributes_init(&attr);
	UA_String nodeDefaultValue;
	nodeDefaultValue.data = (UA_Byte*) defaultNodeValue.c_str();
	nodeDefaultValue.length = defaultNodeValue.length();
	UA_Variant_setScalar(&attr.value, &nodeDefaultValue, &UA_TYPES[UA_TYPES_STRING]);
	attr.description = UA_LOCALIZEDTEXT("en_US", (char*) nodeName.c_str());
	attr.displayName = UA_LOCALIZEDTEXT("en_US", (char*) nodeName.c_str());

	/* 2) Add the variable node to the information model */
	UA_NodeId myIntegerNodeId = UA_NODEID_STRING(0, (char*) nodeName.c_str());
	UA_QualifiedName myIntegerName = UA_QUALIFIEDNAME(0, (char*) nodeName.c_str());
	UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
	UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
	return UA_Server_addVariableNode(server, myIntegerNodeId, parentNodeId, parentReferenceNodeId, myIntegerName, UA_NODEID_NULL, attr, NULL, NULL);
}
```

SvenManier, 2016-07-12:
Tested it with your static_server. Monitored the.answer and then currentTime; result was the same, the.answer made its initial response, but after currentTime was subscribed, the.answer was overwritten und I got no more changes from the node

ichrispa, 2016-07-13:
Hi @SvenManier,

the server polls the values you subscribed to for changes. A changeItem reply is however tagged to a single subscriptions, usually the last one to have a changed item. Have you checked if the server has set the "moreNotifications" flag in the reply? That would indicate that although you only got a reply conerning sub1, sub2 might also contain changes that could not be send.

If so, you should use the RePublish service to get the extra subscription's notification.

Regards,

ichrispa

SvenManier, 2016-07-15:
Where can I look for the flag? In the UA_DataValue I get in the handlingFunction I cant see anything like that

ichrispa, 2016-09-21:
Hi @SvenManier,

You should be able to see the flag when decoding the PublishResponse in Wireshark (or a similar packet capture software).

I have been running the 0.2 rc with multiple subscriptions on various occasions. So far, I cannot confirm that the subscriptions are broken.

Best regards,

ichrispa

jpfr, 2017-04-07:
@ichrispa Are you still onto this?
It's the last blocker for 0.2.

ichrispa, 2017-04-24:
Hi everyone, sorry for the delay.

We have multiple setups the 0.2 branch running with multiple subscriptions and I cannot reproduce @SvenManier's error. To unblock release of the 0.2 branche, I am closing this issue.

closed 2017-04-24 by ichrispa