open62541/open62541 #667: Build PDF documentation for releases

jpfr, 2016-07-06:
Sphinx is generating nice-looking PDFs with the doc_latex make target.

Here's a recent build:
http://www.pdf-archive.com/2016/07/06/open62541/

We should include PDF documentation with the release zips.
Who knows the internals of the release process? @Stasik0 ?

Adding the documentation to the webpage can be done manually once 0.2 is out.
Also, we should stop updating the html documentation on the homepage automatically after 0.2 happens.

Stasik0, 2016-07-07:
docs are already only pushed on updates of 0.2 branch https://github.com/open62541/open62541/blob/0.2/.travis.yml#L96
is the pdf file already generated automatically?

jpfr, 2016-07-07:
> is the pdf file already generated automatically?

No. But should be doable. apt-get texlive and a few lines of script.

jpfr, 2016-11-21:
implemented on 0.2

closed 2016-11-21 by jpfr