open62541/open62541 #668: 0.2 examples cleanup

jpfr, 2016-07-06:
Hey all,

the following changes will eventually land if nobody is against them:

- clean up server.c to **not contain raspi/windows specific temperature** nodes. /examples/server.c is probably the first example that people look at. And it's way too complicated for being exemplary.
- move examples out of the main cmakelists.txt. They can be self-contained in the examples folder.

@ichrispa Is there a way to speed up generation for `example_nodeset.c`? For example by using a smaller nodeset file... I'd like to have all examples behind a common cmake flag. But `example_nodeset.c` takes ages to generate. That might put people off.

Stasik0, 2016-07-07:
yea we can reduce the amount of code in server.c or make a special rpi demo

ichrispa, 2016-07-13:
Hi @jpfr,

an addon request for consideration: Maybe we should restructure the examples folder with its own CMakeList as a project template and generate an example executable of that subproject? A use-case that comes up a lot is "how do I setup my own project?". The examples are all integrated into the main CMakeList and are quite difficult to convert into stand-alone environments. Most of my students just start modifying one of our examples, then get stuck when they need to change the CMake file (too advanced)...

Concerning your question:

Generation times scales with the amount of nodes. If we use the Opc.Ua.Nodeset2.Minimal.xml, I only have 300ms of generation time. As far as I can tell, server_nodeset does not require a full namespace 0. Its as simple as inserting ".Minimal" into the CMake?

From expirience working with userspace namespaces (which is kind of common in my applications): The generation overhead is annoying, but it usually only comes up once in the development process. In particular, you only compile the model initializer when the model changes. So yes, slow, but its a one-time overhead. I also tend to include an include/model_prebuild folder and an ENABLE_BUILDMODEL cmake flag to enable project compilation without generation.

Best regards,

ichrispa

jpfr, 2016-07-14:
Hey @ichrispa ,

Yep, Good points.

How do you like this CMakeLists file in the examples folder?
https://github.com/open62541/open62541/blob/cleanup_cmake/examples/CMakeLists.txt
Adding your own example means copy/pasting two lines.
It needs a bit more work in the build scripts because now the example binaries end up in the /build/examples folder.

I will have a look at how to change to the minimal nodeset. If it is as fast as you say, we keep it activated for the standard build...

jpfr, 2016-07-18:
Fixed in #681.

closed 2016-07-18 by jpfr