open62541/open62541 #672: Generating an OPC UA Information Model from XML not works when use Opc.Ua.Di.NodeSet2.xml

YangXinyu, 2016-07-09:
I have use UA Modeler to create  a "testDiType" as subtype of "TopologyElementType" in OPC.Ua.Di.NodeSet2.xml. Then i export my XML as deep_test.xml . The code in deep_test.xml is:

`<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:s1="http://yourorganisation.org/deep_text/Types.xsd" xmlns:s2="http://opcfoundation.org/UA/DI/Types.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://yourorganisation.org/deep_text/</Uri>
        <Uri>http://opcfoundation.org/UA/DI/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="String">i=12</Alias>
        <Alias Alias="HasModellingRule">i=37</Alias>
        <Alias Alias="HasTypeDefinition">i=40</Alias>
        <Alias Alias="HasSubtype">i=45</Alias>
        <Alias Alias="HasProperty">i=46</Alias>
    </Aliases>
    <Extensions>
        <Extension>
            <ModelInfo Tool="UaModeler" Hash="Wfsl/sNX9XbsFEhc5Yc0Uw==" Version="1.4.4"/>
        </Extension>
    </Extensions>
    <UAObjectType NodeId="ns=1;i=1003" BrowseName="1:testDiType">
        <DisplayName>testDiType</DisplayName>
        <References>
            <Reference ReferenceType="HasProperty">ns=1;i=6002</Reference>
            <Reference ReferenceType="HasSubtype" IsForward="false">ns=2;i=1001</Reference>
        </References>
    </UAObjectType>
    <UAVariable DataType="String" ParentNodeId="ns=1;i=1003" NodeId="ns=1;i=6002" BrowseName="1:id" UserAccessLevel="3" AccessLevel="3">
        <DisplayName>id</DisplayName>
        <References>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=1003</Reference>
        </References>
    </UAVariable>
</UANodeSet>
`
The Problem 1 is, the namespace of testType should be 2, but is 1. And the namespace of TopologyElementType should be 1, but is 2.

Then i change the Cmakelists.txt with the code:

`${PROJECT_SOURCE_DIR}/tools/schema/namespace0/${GENERATE_NAMESPACE0_FILE}                                                                  ${PROJECT_SOURCE_DIR}/tools/schema/namespace1/Opc.Ua.Di.NodeSet2.xml
                                                ${PROJECT_SOURCE_DIR}/tools/schema/namespace1/deep_test.xml                                                                                         
                                                ${PROJECT_BINARY_DIR}/src_generated/ua_namespaceinit_generated` 

Then i build the library use make, but the error commt: 

`[ 10%] Generating src_generated/ua_namespaceinit_generated.c, src_generated/ua_namespaceinit_generated.h
0.0
ns=2;i=1001 ns=2;i=1001
1.0
Traceback (most recent call last):
  File "/Users/yangxinyu/Desktop/DA/code/open62541-0.2.0-RC1-Mac/tools/pyUANamespace/generate_open62541CCode.py", line 125, in <module>
    preProc.preprocessAll()
  File "/Users/yangxinyu/Desktop/DA/code/open62541-0.2.0-RC1-Mac/tools/pyUANamespace/open62541_XMLPreprocessor.py", line 375, in preprocessAll
    self.preprocess_linkDependantModels()
  File "/Users/yangxinyu/Desktop/DA/code/open62541-0.2.0-RC1-Mac/tools/pyUANamespace/open62541_XMLPreprocessor.py", line 350, in preprocess_linkDependantModels
    matches.append(c, tDoc)
TypeError: append() takes exactly one argument (2 given)
make[2]: *** [src_generated/ua_namespaceinit_generated.c] Error 1
make[2]: *** Deleting file `src_generated/ua_namespaceinit_generated.c'
make[1]: *** [CMakeFiles/open62541-object.dir/all] Error 2
make: *** [all] Error 2`

I just do not know, why it can not work when use Opc.Ua.Di.NodeSet2.xml. 

Tank you very much for your help!

YangXinyu, 2016-07-11:
Hey all, 
i have found another problem by "Generating an OPC UA Information Model from XML Descriptions" in tutorial. 

I copy the pump.xml in the tutorial and build the library and create file "ua_namespaceinit_generated.h" with the following definitions :

`#define UA_NS2ID_PROVIDESINPUTTO 4001`
`#define UA_NS2ID_FIELDDEVICE 1001`
`#define UA_NS2ID_PUMP 1002`
`#define UA_NS2ID_STARTPUMP 7001`
`#define UA_NS2ID_STOPPUMP 7002`

But in example the Namespaceindex should be 1:
`#define UA_NS1ID_FIELDDEVICE`

I hava also found, that all the new created node are with the nodeId : ns= 2 but not ns =1.
I just do not know , why it do not generate the Namespaceindex 1 but direct generate the namespaceindex 2.

Thanks for your help !

Stasik0, 2016-07-11:
@ichrispa

markusgraube, 2016-07-13:
@YangXinyu Does this fix work for you?

YangXinyu, 2016-07-13:
@markusgraube 
Ja, es funktioniert jetzt wieder super. Dank sehr !

markusgraube, 2016-07-13:
created a PR #683

closed 2016-07-14 by Stasik0