open62541/open62541 #673: Creating object instances not support inheritance

YangXinyu, 2016-07-11:
Hey all,

I try to run the ` Creating object instances` follow the tutorial. It should create a instance of pump.

After i run the code, the pump was created, but the instance Pump1 has only the property : `MotorRPM, isOn, startPump and stopPump`. 

But the property :`ManufacturerName and ModelName`, which inherited from the father : `FieldDevice`  has not been created. 

Following is my code to create instance:
`UA_Server_addObjectNode(server, UA_NODEID_NUMERIC(1, 196),
		UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
		UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES), UA_QUALIFIEDNAME(1, "Pump1"),
		UA_NODEID_NUMERIC(2, UA_NS2ID_PUMP), object_attr, &theAnswerCallback, &createdNodeId);`

Thanks for your help !

jpfr, 2016-07-11:
Hey @YangXinyu,

can you please activate verbose logs and show us the log output?
Here are some indications on how to state issues so that we can resolve them quickly.
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Thanks!

YangXinyu, 2016-07-11:
The version of open62541 i used is ` v0.2.0-RC1` .

I want to try the example of `Creating object instances` in tutorial: `Generating an OPC UA Information Model from XML Descriptions.` [http://open62541.org/doc/current/tutorial_noderelations.html#](url)

1. I create a pumpType namespace use xml with the structur:
[image]
 
2. I compile the stack use cmake and it create the header file "ua_namespaceinit_generated.h" with the following definitions :
`#define UA_NS2ID_PROVIDESINPUTTO 4001`
`#define UA_NS2ID_FIELDDEVICE 1001`
`#define UA_NS2ID_PUMP 1002`
`#define UA_NS2ID_STARTPUMP 7001`
`#define UA_NS2ID_STOPPUMP 7002`

3. I run the example code: `Creating object instances` and create a OPC UA Server:
```c
UA_Server_addObjectNode(server, UA_NODEID_NUMERIC(1, 196),
		UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
		UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES), UA_QUALIFIEDNAME(1, "Pump1"),
		UA_NODEID_NUMERIC(2, UA_NS2ID_PUMP), object_attr, &theAnswerCallback, &createdNodeId);
```

4.The output when i run the OPC UA Server is:

`Created new node ns=1, i=196 according to template ns=2; i=1002` 

`Created new node ns=1, i=1501 according to template ns=2; i=6003`

`Created new node ns=1, i=1502 according to template ns=2; i=6 004`

5.The problem is: 

The created instance Pump1 has only the property : `MotorRPM, isOn, startPump and stopPump`. 
But the property :`ManufacturerName and ModelName`, which inherited from the father : `FieldDevice`  has not been created. 

Thank you for your help !

ichrispa, 2016-07-13:
Hi @jpfr,

this error occured after you rewrote the instantiation. I reported it to you via IRC, but we deemed it "not important" at the time.

As a point of reference: My old code instantiated the object, then traversed any inverse hasSubtype references and instantiated any objects/variables contained therein. This was only true for hasComponent and hasProperty references. We had a discussion (by phone I think) that this behavior was wrong: any subtype of hasPropery/Component would need to be instantiated as well, which was why we decided not to implement the feature at that time (better not at all than buggy).

Regards,

Chris

jpfr, 2016-07-14:
As we discussed back then, the feature was implemented to also take subtypes of hasPropery/Component into account.

We make a browse inside the variable type (or object type) and instantiate everything we find via an *aggregates* relation or subtype of *aggregates*.

https://github.com/open62541/open62541/blob/0.2/src/server/ua_services_nodemanagement.c#L407

Still trying to figure out what went wrong for @YangXinyu ...

jpfr, 2016-09-13:
Hello @ichrispa ,

you recently added support for multi-inheritance. 0f872fd0c0d4693815efa9f6af28abac377f14c8
Multi-inheritance is a good idea. However, the current use is in contradiction to the OPC UA standard.

> Part 4, § 7.13:
> Each Variable and each Object shall be the SourceNode of exactly one HasTypeDefinition Reference.

It is currently possible (underspecified) to have an ObjectType interhit from several parent ObjectTypes. 
If you are okay with that, I would then make multi-inheritance possible for types, but not instances.

jpfr, 2016-09-13:
@ichrispa 
Aah. It turns out the part you added is only concerned with multi-inheritance at the type level.
So the change only affects code that existed before...

ichrispa, 2016-10-11:
Hi @jpfr,

the multi-inheritance in UA concerns itself with creating an object of different types (ie. A serverType object that is also a StateMachineType is not permitted). I am only implementing inheritance from superTypes, which is imho covered by the spec. I have no aspirations to create multi-typed objects, but detailed inheritance is a pretty helpful (and quite standard) modelling tool.

As far as I can tell, this issue is closed with the latest fixes? @YangXinyu: Reopen if I am wrong.

Best regards,
ichrispa

closed 2016-10-11 by ichrispa

sunwenfeng, 2017-05-03:
I have successfully created the FIELDDEVICE object type, now I try to run the Creating object instances follow the tutorial. I changed my code with reference to your answer. After running(gcc -g -std=c99 -Wl,-rpath,'pwd' -I ./include -L . -DUA_ENABLE_METHODCALLS -o pump_instance ./pump_instance.c -lopen62541), the instance pump1 did not create , can you tell me how you running the code ? What should I do before running?@YangXinyu.Thanks!