open62541/open62541 #675: Local Discovery Server

Pro, 2016-07-12:
We need the concept of a centralized LDS (Local Discovery Server) (see http://www.automation.com/automation-news/article/how-opc-ua-clients-discover-servers-part-3)

Does open62541 support this kind of setup?

The required methods for this are:
- register server
- find servers
- get endpoints

It looks like the last two are already included in open62541:
* https://github.com/open62541/open62541/blob/0.2/src/server/ua_services_discovery.c
* https://github.com/open62541/open62541/blob/0.2/src/client/ua_client.c#L378

I couldn't find the `register server`. Is there any plan to support LDS?

I would be happy to help you add the functionality by creating some example code. Just let me know how you would approach this.

ichrispa, 2016-07-13:
Hi @Pro,

I am currently using a custom UDP broadcast mechanism to implement server discorvery. Nice to see someone's taking up the job of implementing the spec-conformant discovery service.

Best regards,

ichrispa

Pro, 2016-07-18:
The first simple implementation of an LDS supporting `RegisterServer` and `FindServers` is in the current master branch. Not yet supported (but planned) is LDS-ME and GDS.

closed 2016-07-18 by Pro