open62541/open62541 #676: UA_Client_connect do not work

tpiat, 2016-07-12:
Version 1f6c816802, UA_Client_connect returns 0. Client works.
Versions 056cf2f6e5, 751f851778, d3ec67e70b:
UA_Client_connect returns 0x8013000, 

Params is identical.

Where could be the problem?

Stasik0, 2016-07-12:
* which server are you using so we can try? 
* comment out this line: it will probably work https://github.com/open62541/open62541/blob/0.2/src/client/ua_client.c#L288
* set loglevel to 100 and give us the log starting with "The sequence number was not increased by one."

Stasik0, 2016-07-12:
@Torben-D I suppose the other server does not start counting sequence ids with 0
* What number is allowed?
* What increase is allowed (+1) or just strictly monotonic?

tpiat, 2016-07-12:
I use SIMATIC server.

I  have commented all lines with UA_SecureChannel_processSequenceNumber call.
(contain retval = UA_SecureChannel_processSequenceNumber)
Client works now.

What does it mean?

Stasik0, 2016-07-12:
I need an output of the debug code, can you just print

seqHeader.sequenceNumber
and
client->channel->receiveSequenceNumber + 1

somehow

Torben-D, 2016-07-12:
@Stasik0 :
- The allowed number is the old number +1. 
- At the first open request the sending number of the server is the starting number for the receiveNumber of the client.

Mabye the server does not increase the number?

Stasik0, 2016-07-12:
@tpiat could you confirm the fix please

tpiat, 2016-07-13:
It Works.

closed 2016-07-13 by tpiat