open62541/open62541 #680: ns0 generation does not work anymore

FlorianPalm, 2016-07-13:
The following error occurs if the stack should be generated with UA_ENABLE_GENERATE_NAMESPACE0=ON option:

``
[  7%] Generating src_generated/ua_namespaceinit_generated.c, src_generated/ua_namespaceinit_generated.h
INFO:__main__:Preprocessing /home/opcua/open62541_clean/tools/schema/namespace0/Opc.Ua.NodeSet2.xml
INFO:__main__:Parsing /tmp/Opc.Ua.NodeSet2.xml_preProcessed-Ks3wtL.xml
INFO:__main__:Linking namespace nodes and references
INFO:__main__:Building datatype encoding rules
INFO:__main__:Allocating variables
WARNING:__main__:Can't ignore node, Namespace does currently not contain a node with id i=2292
WARNING:__main__:Can't ignore node, Namespace does currently not contain a node with id i=274
INFO:__main__:Generating Header
Traceback (most recent call last):
  File "/home/opcua/open62541_clean/tools/pyUANamespace/generate_open62541CCode.py", line 176, in <module>
    generatedCode=ns.printOpen62541Header(ignoreNodes, args.suppressedAttributes, outfilename=basename(args.outputFile))
  File "/home/opcua/open62541_clean/tools/pyUANamespace/ua_namespace.py", line 697, in printOpen62541Header
    code = code + n.printOpen62541CCode(unPrintedNodes, unPrintedRefs, supressGenerationOfAttribute=supressGenerationOfAttribute)
  File "/home/opcua/open62541_clean/tools/pyUANamespace/ua_node_types.py", line 679, in printOpen62541CCode
    code = code + codegen.getCreateNodeNoBootstrap(self, parentNode, parentRef, unPrintedNodes)
  File "/home/opcua/open62541_clean/tools/pyUANamespace/open62541_MacroHelper.py", line 152, in getCreateNodeNoBootstrap
    if r.target().value() != None:
AttributeError: opcua_node_object_t instance has no attribute 'value'
make[2]: *** [src_generated/ua_namespaceinit_generated.c] Error 1
make[1]: *** [CMakeFiles/open62541-object.dir/all] Error 2
make: *** [all] Error 2
``

Stasik0, 2016-07-13:
correction: GENERATE_NAMESPACE0_FILE=Opc.Ua.NodeSet2.Minimal.xml works but GENERATE_NAMESPACE0_FILE=Opc.Ua.NodeSet2.xml crashes

ichrispa, 2016-07-13:
very minor bug; already fixed locally... please stand by

ichrispa, 2016-07-13:
On second thought... it appears someone has somewhat modified the script to some extend since I last peeked in there. I reverted the commit (sry for the inconvinience).

closed 2016-07-13 by ichrispa

ichrispa, 2016-07-13:
@FlorianPalm: Please check the crrent 0.2 version - if the error persists, reopen this issue please.

commit a3164aca (2016-07-13): Fixed namespace compiler preprocessor implicitly expecting values in non-variable node types (fixes #680)
commit ec853c3b (2016-07-13): Revert "Fixed namespace compiler preprocessor implicitly expecting values in non-variable node types (fixes #680)"

This reverts commit a3164aca6bf8b519da3e43eb54e8b2436ae63795.
commit 96a3f41c (2016-07-13): Fixed indentation error in XML Preprocessor (fixes #680)
commit 1b71de44 (2016-08-01): Revert "Fixed namespace compiler preprocessor implicitly expecting values in non-variable node types (fixes #680)"

This reverts commit a3164aca6bf8b519da3e43eb54e8b2436ae63795.