open62541/open62541 #686: DataValue parsing incorrect when source picoseconds or server pico seconds present.

prateekbansal, 2016-07-14:
I tried to parse `DataValue` using this data:

```
// { 0x3F,
//                           0x01, 0x00, // Variant Value
//                           0x64, 0x00, 0x00, 0x00, // StatusCode
//                           0x00, 0x64, 0xDF, 0x3E, 0x6E, 0xDB, 0x52, 0x06, // sourceTimestamp = 455667778865554432
//                           0xE0, 0x15, //  sourcePicoSeconds = 5600
//                           0x00, 0x64, 0x69, 0x9C, 0xE6, 0x20, 0xB6, 0x07, // serverTimestamp = 555667778865554432
//                           0xA8, 0x16 //  serverPicoSeconds = 5800
// }
```
However, the output that I get is:
```
  {"hasServerPicoseconds":true,"hasServerTimestamp":true
  ,"hasSourcePicoseconds":true,"hasSourceTimestamp":true,"hasStatus":true
  ,"hasValue":true,"serverPicoseconds":999
  ,"serverTimestamp":2370754230320371200,"sourcePicoseconds":999
  ,"sourceTimestamp":455667778865554432,"status":100,"value":false}
```

The `sourceTimeStamp` field is parsed correctly, but any field after and including sourcePicoSeconds is incorrect.

Stasik0, 2016-07-14:
Hi, where does the binary data coming from?

closed 2016-07-14 by jpfr

prateekbansal, 2016-07-14:
This data is from my unit test and could possibly be incorrect. I based my
unit tests off specification at
http://www.commsvr.com/UAModelDesigner/html/7f816c6d-3aa2-4f38-9992-16d892e6fe4a.htm#DataValue

prateekbansal, 2016-07-14:
Part of the issue is fixed. However, there is one more issue: The value of `serverPicoseconds` and `sourcePicoseconds` is capped at 999. As per http://www.commsvr.com/UAModelDesigner/html/7f816c6d-3aa2-4f38-9992-16d892e6fe4a.htm#DataValue, the value could be max 9999.

As per latest run of the test, this is the output that I get:
```
{"hasServerPicoseconds":true,"hasServerTimestamp":true
  ,"hasSourcePicoseconds":true,"hasSourceTimestamp":true,"hasStatus":true
  ,"hasValue":true,"serverPicoseconds":999,"serverTimestamp":555667778865554432
  ,"sourcePicoseconds":999,"sourceTimestamp":455667778865554432,"status":100
  ,"value":false}
```

prateekbansal, 2016-07-14:
Opened issue #692 to track this further.

commit cad923d7 (2016-07-14): fix #686