open62541/open62541 #688: Session not based on UA_DateTime_nowMonotonic

NetoLe, 2016-07-14:
Hello!

For me a problem occured, because I am using a non-RTC-device, which get's synchronized with realtime at one point. this leads to a lost session, because its timestamp is way older than the next UA_DateTime_now(), which it calls (used in UA_SessionManager_getSession() -> if(UA_DateTime_now() > current->session.validTill) ).
I'm not totally into the spec, but is the session timeout not just an internal handling and doesn't need to be coupled to the real time? 
I would prefer using UA_DateTime_nowMonotonic() instead.

What do you think?

best regards

jpfr, 2016-07-14:
Good point! See PR #691.

jpfr, 2016-07-14:
Implemented, passed CTT and unit tests.

closed 2016-07-14 by jpfr