open62541/open62541 #689: current version not working with TeslaSCADA?

anyn99, 2016-07-14:
Tested the server example and compiled the simple Tutorial examples for server.
None of those seem to work with the current release of TeslaSCADA.
(www.teslascada.com)
The servers keep complaining about "Unknown requests".

Is this a known issue? Would be nice if TeslaSCADA and open62541 would be compatible i think!

Stasik0, 2016-07-14:
issue is not known. Did you activate subscriptions on the server side? which version are you using?

jpfr, 2016-07-14:
That is probably the client trying to use subscriptions.
If they are not activated, then the server does not know about the service and rejects the requests.

By the way, subscriptions are not automated per default in v0.2-rc1 but will be in the final release.

anyn99, 2016-07-14:
Subscriptions are activated when building with cmake.
They work as expected with UAExpert...
Meanwhile i think it has to do with the secure connection part... i.e. got
an "Unknown request 450", which is something like secure connection
close...But i connect as anonymous.

jpfr, 2016-07-14:
Then the teslascada client is at fault.
450 is the id of CloseSecureChannelRequest.

The log output you see is generated when processing a MSG chunk.
But CloseSecureChannel should only be received with a CLO chunk.

anyn99, 2016-07-14:
I am compiling open2541 v0.2.0-RC1-189-gcad923d.

when trying this simple example:
http://open62541.org/doc/current/tutorial_server_variables.html

the server hangs when connecting to TeslaSCADA:
[07/14/2016 16:51:43.496] info/network  Connection 4 | New connection over TCP from 192.168.xxx.xxx:49428

Atm i am not completly sure if it also could be an network issue, but it works perfectly when connecting to UAExpert on the same setup. Couldn't reproduce this error with id 450. Will investigate further...

anyn99, 2016-07-14:
Got it reproduced! Sometimes it just fails like in my last comment, but sometimes it fails like this:

[07/14/2016 17:01:25.802] info/network  Connection 4 | New connection over TCP from 192.168.xxx.xxx:49467
[07/14/2016 17:01:26.102] trace/network Connection 4 | Process a HEL
[07/14/2016 17:01:26.106] trace/network Connection 4 | Process a OPN
[07/14/2016 17:01:26.106] info/channel  Connection 4 | SecureChannel 1 | OpenSecureChannel: Opened SecureChannel
[07/14/2016 17:01:26.109] trace/network Connection 4 | Process a MSG
[07/14/2016 17:01:26.109] trace/channel Connection 4 | SecureChannel 1 | Final chunk message
[07/14/2016 17:01:26.109] debug/session Try to use Session with token {00000000-0000-0000-0000-000000000000} but is not found
[07/14/2016 17:01:26.109] debug/session Connection 4 | SecureChannel 1 | Session {00000000-0000-0000-0000-000000000000} | Processing GetEndpointsRequest with an empty endpointUrl
[07/14/2016 17:01:26.113] trace/network Connection 4 | Process a MSG
[07/14/2016 17:01:26.113] trace/channel Connection 4 | SecureChannel 1 | Final chunk message
[07/14/2016 17:01:26.113] info/channel  Connection 4 | SecureChannel 1 | Unknown request 450
[07/14/2016 17:01:26.114] info/network  Connection 4 | Connection closed from remote
[07/14/2016 17:01:26.261] info/network  Connection 4 | New connection over TCP from 192.168.xxx.xxx:49468
[07/14/2016 17:01:33.855] debug/channel Connection 0 | SecureChannel 1 | SecureChannel has timed out
[07/14/2016 17:02:26.232] info/network  Connection 4 | Connection closed from remote
[07/14/2016 17:02:26.234] info/network  Connection 4 | New connection over TCP from 192.168.xxx.xxx:49469

I really hope its TeslaSCADAs fault... does anyone happen to know a freely available SCADA which is known to work well with open62541?

jpfr, 2016-07-14:
Hey @anyn99,

we should get it working with TeslaSCADA.
Either by changing our implementation or by filing a bug for TeslaSCADA.

Anyway, the hang sounds bad. Can you reproduce with the current 0.2 branch?
There were a lot of improvements since the v0.2-rc1.

Stasik0, 2016-07-18:
@jpfr he is already testing with https://github.com/open62541/open62541/commit/cad923d719eba45a6f02c86033c10f1cdb1fac7b

jpfr, 2016-07-19:
```
[07/14/2016 17:01:26.113] trace/network Connection 4 | Process a MSG
[07/14/2016 17:01:26.113] trace/channel Connection 4 | SecureChannel 1 | Final chunk message
[07/14/2016 17:01:26.113] info/channel Connection 4 | SecureChannel 1 | Unknown request 450
```

That is not what I expect to be happening. Closing the securechannel should be a `CLO` message and not a `MSG` message.

Stasik0, 2016-07-19:
This seem to be a bug in TeslaSCADA, apparently they use an opcf java stack.

Stasik0, 2016-07-19:
did you see #707? you need a fresh source and a server certificate that is loaded in the server e.g. in server_certificate example

anyn99, 2016-07-20:
The most recent version ( 5d841cb ) is now even killing my open62541-Server!:

last logs:
Connection 33 | New connection over TCP from 192.168.170.111:50298
Connection 33 | Process a HEL

after that my binary crashes, here some backtrace:
*** Error in `./xxx': free(): invalid next size (fast): 0x00007f6044027d70 ***
libopen62541.so(UA_deleteMembers+0xc8) [0x7f606454c3d8]
open62541/build/libopen62541.so(+0x28e9d) [0x7f606456ce9d]
(memory allocation issue??)

I admit, i haven't loaded a server certificate yet. But even so it should not crash completely, right?

Stasik0, 2016-07-20:
right

Stasik0, 2016-07-21:
@anyn99 should work with the latest fix, in order to get connected, make sure a certificate is there (enable UA_BUILD_SELFSIGNED_CERTIFICATE) and start server_certificate

anyn99, 2016-07-22:
Hi again!
Im testing with TeslaSCADA 2.06.1.
Included the self-signed certificate the same way as in server-certificate example.
with newest fix the server is not crashing anymore.

But it is still not possible to access the server, maybe i missed something?
Here's the log:
`INFO: [NETWORK]: CONNECTION 36 | NEW CONNECTION OVER TCP FROM 192.168.170.111:49753
TRACE: [NETWORK]: CONNECTION 36 | PROCESS A HEL
TRACE: [NETWORK]: CONNECTION 36 | PROCESS A OPN
INFO: [SECURECHANNEL]: CONNECTION 36 | SECURECHANNEL 12 | OPENSECURECHANNEL: OPENED SECURECHANNEL
TRACE: [NETWORK]: CONNECTION 36 | PROCESS A MSG
TRACE: [SECURECHANNEL]: CONNECTION 36 | SECURECHANNEL 12 | FINAL CHUNK MESSAGE
INFO: [DEBUG][SESSION]: TRY TO USE SESSION WITH TOKEN {00000000-0000-0000-0000-000000000000} BUT IS NOT FOUND
INFO: [DEBUG][SESSION]: CONNECTION 36 | SECURECHANNEL 12 | SESSION {00000000-0000-0000-0000-000000000000} | PROCESSING GETENDPOINTSREQUEST WITH AN EMPTY ENDPOINTURL
TRACE: [NETWORK]: CONNECTION 36 | PROCESS A MSG
TRACE: [SECURECHANNEL]: CONNECTION 36 | SECURECHANNEL 12 | FINAL CHUNK MESSAGE
INFO: [SECURECHANNEL]: CONNECTION 36 | SECURECHANNEL 12 | UNKNOWN REQUEST 450
INFO: [NETWORK]: CONNECTION 36 | CONNECTION CLOSED FROM REMOTE
INFO: [NETWORK]: CONNECTION 36 | NEW CONNECTION OVER TCP FROM 192.168.170.111:49754
TRACE: [NETWORK]: CONNECTION 36 | PROCESS A HEL
TRACE: [NETWORK]: CONNECTION 36 | PROCESS A OPN
INFO: [SECURECHANNEL]: CONNECTION 36 | SECURECHANNEL 13 | OPENSECURECHANNEL: OPENED SECURECHANNEL
TRACE: [NETWORK]: CONNECTION 36 | PROCESS A MSG
TRACE: [SECURECHANNEL]: CONNECTION 36 | SECURECHANNEL 13 | FINAL CHUNK MESSAGE
INFO: [DEBUG][SECURECHANNEL]: CONNECTION 36 | SECURECHANNEL 13 | SESSION {12423BB8-3C36-BC69-A10A-30438E48B4EB} CREATED
INFO: [DEBUG][SECURECHANNEL]: CONNECTION 0 | SECURECHANNEL 12 | SECURECHANNEL HAS TIMED OUT
TRACE: [NETWORK]: CONNECTION 32 | PROCESS A OPN
INFO: [DEBUG][SECURECHANNEL]: CONNECTION 32 | SECURECHANNEL 2 | OPENSECURECHANNEL: SECURECHANNEL RENEWED                                                             
TRACE: [NETWORK]: CONNECTION 33 | PROCESS A OPN                                                                                                                       
INFO: [DEBUG][SECURECHANNEL]: CONNECTION 33 | SECURECHANNEL 4 | OPENSECURECHANNEL: SECURECHANNEL RENEWED                                                             
TRACE: [NETWORK]: CONNECTION 34 | PROCESS A OPN                                                                                                                       
INFO: [DEBUG][SECURECHANNEL]: CONNECTION 34 | SECURECHANNEL 6 | OPENSECURECHANNEL: SECURECHANNEL RENEWED                                                             
TRACE: [NETWORK]: CONNECTION 35 | PROCESS A OPN                                                                                                                       
INFO: [DEBUG][SECURECHANNEL]: CONNECTION 35 | SECURECHANNEL 8 | OPENSECURECHANNEL: SECURECHANNEL RENEWED`                       

Anyway im testing now with "Ignition" from Inductive Automation, its working like a charm!

Stasik0, 2016-07-22:
@anyn99 I tested it yesterday with the TeslaScada and it was working, is you server certificate loaded? Did it worked with a "server_certificate" example? Is it showing "certificate" loaded message?

Stasik0, 2016-07-22:
@anyn99 maybe we can hook-up in IRC

Stasik0, 2016-07-22:
the trace is actually looking good

anyn99, 2016-07-22:
hi there! 

My bad! Tried to load the certificate just after creating the server with UA_Server_new.
TeslaSCADA and open62541 are talking with each other now! 

btw, just created an open62541 package for OpenWRT, works like a charm on an embedded device here.

Thanks a lot for this wonderful piece of software!
best wishes,
Daniel

Stasik0, 2016-07-22:
very good! is the OpenWRT package available?

closed 2016-07-22 by Stasik0

anyn99, 2016-07-26:
yes I want to release it soon  :-) Do you know a community site where others can find it? Or maybe i just use github?