open62541/open62541 #693: Variants with Array of Bytes should throw error

prateekbansal, 2016-07-14:
Input (array of 3 bytes): 
```
0x83, 0x03, 0x00, 0x00, 0x00, 0xD6, 0x2A, 0x2B
```

Expected: Error. Instead the bytes are parsed successfully.

jpfr, 2016-07-14:
Why did you expect an error?

- `0x83`: Array (0x80) of Byte (0xx03)
- `0x03, 0x00, 0x00, 0x00`: Array contains three elements
- `0xD6, 0x2A, 0x2B`: thre bytes

Looks fine to me.

prateekbansal, 2016-07-14:
As per http://www.commsvr.com/UAModelDesigner/html/7f816c6d-3aa2-4f38-9992-16d892e6fe4a.htm#DataValue

It says "Array of Bytes are not permitted"

prateekbansal, 2016-07-14:
Also, the "Value shall not be Variant" restriction is that. But I didn't test it.

jpfr, 2016-07-14:
>  It says "Array of Bytes are not permitted"

The current version of the spec (v1.03) changed this to

> Many implementation platforms do not distinguish between one dimensional
Arrays of Bytes and ByteStrings. For this reason, decoders are allowed to
automatically convert an Array of Bytes to a ByteString.

Your second point:
> Also, the "Value shall not be Variant" restriction is there. But I didn't test it.

True. We do not prevent you to pack variants inside a variant.
I'd leave this as **wontfix** since the implementation is easier like that and it is unlikely to cause problems.
Anybody wants to really really have that feature?

closed 2016-07-19 by jpfr