open62541/open62541 #701: mDNS library for LDS-ME

Pro, 2016-07-18:
I'm currently looking into LDS-ME (Local Discovery Server with Multicast Extension).
This requires mDNS and DNS-SD support to automatically detect other LDS servers in the current Multicast Network (similar to zeroconf).

For that we need a mDNS library or source code which we can integrate into open62541.
I've looked around and found a few libraries:

- [mDNSResponder](https://github.com/Apple-FOSS-Mirror/mDNSResponder/) - Apache License 2.0 - Most comprehensive and portable implementation. See also https://opensource.apple.com/source/mDNSResponder/mDNSResponder-625.41.2/
- [avahi](https://github.com/lathiat/avahi) - LGPL - mainly used in Linux, comparable with Bonjour

- [mdns-lib](https://github.com/rampantpixels/mdns_lib) - Public domain - has some dependencies with other public domain library, which would need to be rewritten
- [mdnsd](https://github.com/troglobit/mdnsd) - 3-clause BSD - not that popular compared to others

Before I start to implement LDS-ME using one of the above libraries, I would like to get some feedback from you (@Stasik0 @jpfr).

Which one would you prefer?
I prefer mDNSResponder or avahi.

Of course it would also be possible to implement everything ourselves, but this would be a huge amount of work and not really necessary.

Any thoughts?

jpfr, 2016-07-18:
Hey Stefan,

I cannot speak for @Stasik0 and the others. My criteria for the design would contain the following.

## Overall structure of open62541
- The main library (everything in /src and /include) builds with raw C99. All extensions (also POSIX) are encapsulated in the /plugins directory.
- External dependencies go into the /deps directory. We can also have git submodules pointing to an external repository/commit that is loaded when required.
- It should be possible to merge all input files into a single pair of .c/.h files. Including files that come from /deps.

## Control flow
You can add the mdns functionality as either a repeated job with a short enough interval or into the main-loop that also queries the normal network layers for messages. Then we don't have to worry about (multi-) threading etc.

## Licenses
We promise users a library they can bake into their projekt without making their own code open source. That includes static linking. We cannot do that with avahi (raw LGPL).

## Dependencies
I (and maybe others) have a strong bias towards small and self-contained dependencies.
Ideally, we can use our own networking if (at some future time) required and push messages directly into the mdns processing layer.
mDNSResponder is huge and makes a lot of assumptions regarding available libraries / OS / networking stack. We would also get a lexer as part of our build chain.

mdnsd (https://github.com/troglobit/mdnsd), openmdns (https://github.com/haesbaert/mdnsd), and tinysvcmdns (https://bitbucket.org/geekman/tinysvcmdns/overview) all look small and viable.

Do we need **everything** that mdns does or just a subset of features?

Stasik0, 2016-07-18:
+1 for @jpfr's arguments I could not express it better

Pro, 2016-07-18:
Sounds very good for me too. I'm also a fan of self-containing dependencies.
For LDS-ME we only need a basic mDNS responder daemon which returns the server information, and a client which is able to query this information.

I'll have a look at the smaller libraries and try to come up with a good implementation.

Can you create a separate feature branch (e.g. `feature/discovery`), based on master, where I can submit my pull requests. For now I think it would be better if it is separated, until discovery (LDS-ME and GDS) is implemented completely.

Stasik0, 2016-07-18:
@Pro cannot you work on one of your fork's branches? Just curious.

Pro, 2016-07-18:
Yes, I can, and I currently do. As soon as my first implementation is ready, I would like to create a PR into a separate branch, so you can contribute/finetune my changes, until it is ready to merge it into master. But it's also no problem for me, to just code in my fork's branch.

jpfr, 2016-07-18:
> I would like to create a PR into a separate branch, so you can contribute/finetune my changes, until it is ready to merge it into master.

Good idea. Let's do it like that.
For small things that come up during codee reviews it's often easier to just change it right away in a common branch.
https://github.com/open62541/open62541/tree/feature_mdns

Pro, 2017-02-20:
Discussion continues on PR #732

closed 2017-02-20 by Pro